cesanta/mongoose-os #437: ESP32 Error: flash size is not specified and could not be detected: failed to get flash chip id: error reading response packet: invalid SLIP starting byte: 0x00

boonsanti, 2018-07-12:
$ mos flash -v 4 --logtostderr
```
Loaded demo-c/esp32 version 1.0 (20180712-083735/???)
I0712 16:35:03.103711   13595 ourutil.go:16] Loaded demo-c/esp32 version 1.0 (20180712-083735/???)
Using port /dev/ttyUSB1
I0712 16:35:03.104219   13595 ourutil.go:16] Using port /dev/ttyUSB1
I0712 16:35:03.104240   13595 flash_params.go:104]  -> -1, -1, -1
Opening /dev/ttyUSB1 @ 115200...
I0712 16:35:03.104275   13595 report.go:13] Opening /dev/ttyUSB1 @ 115200...
Connecting to ESP32 ROM, attempt 1 of 10...
I0712 16:35:03.105052   13595 report.go:13] Connecting to ESP32 ROM, attempt 1 of 10...
I0712 16:35:04.705724   13595 rom_client.go:180] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0712 16:35:04.705849   13595 slip.go:90] => (44) 0008240000000000070712205555555555555555555555555555555555555555...
I0712 16:35:04.906939   13595 rom_client.go:180] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0712 16:35:04.907009   13595 slip.go:90] => (44) 0008240000000000070712205555555555555555555555555555555555555555...
I0712 16:35:05.110790   13595 rom_client.go:153] Sync #1 failed: error reading sync response #1: error reading response: error reading: EOF
Connecting to ESP32 ROM, attempt 2 of 10...
I0712 16:35:05.110900   13595 report.go:13] Connecting to ESP32 ROM, attempt 2 of 10...
I0712 16:35:06.711555   13595 rom_client.go:180] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0712 16:35:06.711708   13595 slip.go:90] => (44) 0008240000000000070712205555555555555555555555555555555555555555...
I0712 16:35:06.914816   13595 rom_client.go:180] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0712 16:35:06.914885   13595 slip.go:90] => (44) 0008240000000000070712205555555555555555555555555555555555555555...
I0712 16:35:06.920325   13595 slip.go:46] <= (12) 010804001255555500000000
I0712 16:35:06.920375   13595 rom_client.go:224] <= {cmd:Sync(8) value:1431655698 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.921543   13595 slip.go:46] <= (12) 010804001255555500000000
I0712 16:35:06.921581   13595 rom_client.go:224] <= {cmd:Sync(8) value:1431655698 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.922830   13595 slip.go:46] <= (12) 010804001255555500000000
I0712 16:35:06.922878   13595 rom_client.go:224] <= {cmd:Sync(8) value:1431655698 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.924036   13595 slip.go:46] <= (12) 010804001255555500000000
I0712 16:35:06.924066   13595 rom_client.go:224] <= {cmd:Sync(8) value:1431655698 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.925284   13595 slip.go:46] <= (12) 010804001255555500000000
I0712 16:35:06.925336   13595 rom_client.go:224] <= {cmd:Sync(8) value:1431655698 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.926527   13595 slip.go:46] <= (12) 010804001255555500000000
I0712 16:35:06.926576   13595 rom_client.go:224] <= {cmd:Sync(8) value:1431655698 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.927736   13595 slip.go:46] <= (12) 010804001255555500000000
I0712 16:35:06.927768   13595 rom_client.go:224] <= {cmd:Sync(8) value:1431655698 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.928975   13595 slip.go:46] <= (12) 010804001255555500000000
I0712 16:35:06.929010   13595 rom_client.go:224] <= {cmd:Sync(8) value:1431655698 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.929065   13595 rom_client.go:180] => {cmd:ReadReg(10) csum:0 arg(4):"04a10160"}
I0712 16:35:06.929094   13595 slip.go:90] => (12) 000a04000000000004a10160
I0712 16:35:06.932520   13595 slip.go:46] <= (12) 010a04000000000000000000
I0712 16:35:06.932562   13595 rom_client.go:224] <= {cmd:ReadReg(10) value:0 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.932587   13595 rom_client.go:435] ReadReg(0x6001a104) => 0x00000000
I0712 16:35:06.932605   13595 efuses.go:410] 00000000
I0712 16:35:06.932635   13595 rom_client.go:180] => {cmd:WriteReg(9) csum:0 arg(16):"fca00160a55a0000ffffffff00000000"}
I0712 16:35:06.932678   13595 slip.go:90] => (24) 0009100000000000fca00160a55a0000ffffffff00000000
I0712 16:35:06.936330   13595 slip.go:46] <= (12) 010904000000000000000000
I0712 16:35:06.936371   13595 rom_client.go:224] <= {cmd:WriteReg(9) value:0 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.936399   13595 rom_client.go:449] WriteReg(0x6001a0fc, 0x00005aa5)
I0712 16:35:06.936471   13595 rom_client.go:180] => {cmd:WriteReg(9) csum:0 arg(16):"04a1016001000000ffffffff00000000"}
I0712 16:35:06.936507   13595 slip.go:90] => (24) 000910000000000004a1016001000000ffffffff00000000
I0712 16:35:06.940263   13595 slip.go:46] <= (12) 010904000000000000000000
I0712 16:35:06.940326   13595 rom_client.go:224] <= {cmd:WriteReg(9) value:0 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.940373   13595 rom_client.go:449] WriteReg(0x6001a104, 0x00000001)
I0712 16:35:06.940431   13595 rom_client.go:180] => {cmd:ReadReg(10) csum:0 arg(4):"04a10160"}
I0712 16:35:06.940474   13595 slip.go:90] => (12) 000a04000000000004a10160
I0712 16:35:06.943731   13595 slip.go:46] <= (12) 010a04000000000000000000
I0712 16:35:06.943788   13595 rom_client.go:224] <= {cmd:ReadReg(10) value:0 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.943826   13595 rom_client.go:435] ReadReg(0x6001a104) => 0x00000000
I0712 16:35:06.943850   13595 efuses.go:410] 00000000
I0712 16:35:06.943870   13595 efuses.go:438] eFuseCtlOpRead successful
I0712 16:35:06.943903   13595 efuses.go:120] Reading eFuse block 0 @ 0x6001a000...
I0712 16:35:06.943948   13595 rom_client.go:180] => {cmd:ReadReg(10) csum:0 arg(4):"00a00160"}
I0712 16:35:06.943981   13595 slip.go:90] => (12) 000a04000000000000a00160
I0712 16:35:06.947256   13595 slip.go:46] <= (12) 010a04000000000000000000
I0712 16:35:06.947315   13595 rom_client.go:224] <= {cmd:ReadReg(10) value:0 ok:true lastError:0 body(4):"00000000"}
I0712 16:35:06.947354   13595 rom_client.go:435] ReadReg(0x6001a000) => 0x00000000
I0712 16:35:06.947418   13595 rom_client.go:180] => {cmd:ReadReg(10) csum:0 arg(4):"04a00160"}
I0712 16:35:06.947443   13595 slip.go:90] => (12) 000a04000000000004a00160
I0712 16:35:06.950741   13595 slip.go:46] <= (12) 010a0400545a09c400000000
[... 231 lines cut ...]
/build/mos-4FVQ9m/mos-2.3+7ca03a2~xenial0/obj-x86_64-linux-gnu/src/cesanta.com/mos/main.go:145: 
Error: flash size is not specified and could not be detected: failed to get flash chip id: error reading response packet: invalid SLIP starting byte: 0x00
```

$ esptool.py flash_id
```
esptool.py v2.4.1
Found 1 serial ports
Serial port /dev/ttyUSB1
Connecting........_____.....____
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
MAC: 24:0a:c4:09:5a:54
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

Flash on Arduino OK.
Flash on esptool.py OK.

JamesKingdon, 2019-01-05:
Just hit the same problem with a newly ordered esp32. mos flash reports the chip is ESP32D0WDQ6 R1. Any work-arounds?

Board purchased was:
https://www.ebay.ca/itm/ESP32-ESP32S-CP2102-Development-Board-2-4GHz-Dual-Mode-WiFi-Bluetooth-Antenna/132430583954

JamesKingdon, 2019-01-05:
Ok, some more experimenting and testing with arduino IDE suggests that the issue is with not being able to automatically put the board into the right mode for flashing. If I hold down the 'boot' button during the flashing process things work as expected. Guess I picked a poor choice of dev board.

boonsanti, 2019-02-10:
OK, I find out the problem. It's my usb to ttl poor quality.

closed 2019-02-10 by boonsanti