cesanta/mongoose-os #440: undefined reference to `mgos_upd_set_event_cb'

gpepe, 2018-07-30:
mgos/esp32-build    3.0-r11  

build/objs/mosapp/libmosapp.a(main.o):(.literal.mgos_app_init+0x18): undefined reference to `mgos_upd_set_event_cb'

build/gen/mgos_config.h:897: undefined reference to `mgos_upd_set_event_cb'

#include "mgos.h"
#include "mgos_updater.h"

bool upd_cb(enum mgos_upd_event ev, const void *ev_arg, void *cb_arg) {
...
}

enum mgos_app_init_result mgos_app_init(void) {
  mgos_upd_set_event_cb(upd_cb, nullptr);
...

cpq, 2018-07-30:
please move to the newer events api
`mgos_event_add_handler(MGOS_EVENT_OTA_STATUS, status_cb, NULL);`

astout, 2018-08-13:
Updated to mos 2.4.3 and received same compile error: `undefined reference to 'mgos_upd_set_event_cb'`

I followed @cpq's recommendation. It compiles, but now OTAs do not fire events.

### `mos 2.4.3`

*prior to OTA trigger:*

```c
if (mgos_event_add_handler(MGOS_EVENT_OTA_STATUS, _fw_update_event, NULL)) { ... }
```

*event definition:*

```c
/**
 * Firmware Update Event Handler
 */ 
static void _fw_update_event(int ev, void *ev_arg, void *cb_arg) {
  switch (ev) {
    case MGOS_UPD_EV_INIT: {
      LOG(LL_INFO, ("Starting ESP Firmware Update"));
      return true;
    }
    case MGOS_UPD_EV_BEGIN:
    ...
  }
...
}
```

### `mos 2.2`

*prior to OTA trigger*

```c
mgos_upd_set_event_cb(_fw_update_event, NULL);
```

*event definition:*

```c
/**
 * Firmware Update Event Handler
 */ 
static bool _fw_update_event(enum mgos_upd_event ev, const void *ev_arg, void *cb_arg) {
  switch (ev) {
    case MGOS_UPD_EV_INIT: {
      LOG(LL_INFO, ("Starting ESP Firmware Update"));
      return true;
    }
    case MGOS_UPD_EV_BEGIN:
     ...
  }
...
}
```

In `mos 2.2` the `_fw_update_event` fired fine. What am I missing?

cpq, 2018-08-14:
That API is deprecated in favor of generic events API.
Event fired: `MGOS_EVENT_OTA_STATUS `, see https://github.com/mongoose-os-libs/ota-common/blob/master/src/mgos_updater_common.c#L101
Event data: https://github.com/mongoose-os-libs/ota-common/blob/master/include/mgos_updater_common.h#L30-L35

closed 2018-08-14 by cpq