cesanta/mongoose-os #442: STM32 build issue - cloud

osterwood, 2018-08-09:
When trying build a blank / default firmware image via the cloud docker image, I get the following error:

    Error: neither sources nor prebuilt binary exists for the lib "mongoose" (or, if a library doesn't have any code by design, its mos.yml shouldn't contain "sources")

This was reported back on May 1st, here : https://github.com/mongoose-os-libs/mongoose/issues/1

rojer, 2018-08-20:
yes, we do not provide binary libs for stm32 yet, it's a work in progress and will be addressed before STM32 support is GA

osterwood, 2018-08-20:
Good to know.  I think it would be useful to be more clear to folks that STM32 is WIP.  Your documentation gives the impression that it is supported (without any caveats).

rojer, 2018-08-22:
it took a bit longer than expected, but this problem has been fixed in 2.5.1 (just released)

closed 2018-08-22 by rojer