cesanta/mongoose-os #443: STM32 build issue - local

osterwood, 2018-08-10:
After setting up mos.yml build_vars to define PLATFORM and DEVICE, `most build --local` fails with the following output:

    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/LwIP/src/core/ipv6/ethip6.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/LwIP/src/core/ipv6/icmp6.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/LwIP/src/netif/ethernet.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/FreeRTOS/Source/queue.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/FreeRTOS/Source/croutine.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/FreeRTOS/Source/list.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/FreeRTOS/Source/event_groups.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/FreeRTOS/Source/tasks.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/FreeRTOS/Source/timers.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1/port.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/FreeRTOS/Source/portable/MemMang/heap_3.c
    CC    /opt/STM32Cube_FW_F7_V1.11.0/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.c
    CP    /vagrant/mos-test/stm32-test/deps/ca-bundle/fs/ca.pem -> /vagrant/mos-test/stm32-test/build/fs
    CP    /vagrant/mos-test/stm32-test/fs/index.html -> /vagrant/mos-test/stm32-test/build/fs
    CP    /vagrant/mos-test/stm32-test/build/gen/conf0.json -> /vagrant/mos-test/stm32-test/build/fs
    MKFS  /usr/local/bin/mkspiffs8 98304 {bs: 4096, ps: 128, es: 4096} /vagrant/mos-test/stm32-test/build/fs -> /vagrant/mos-test/stm32-test/build/objs/fw_temp/fs.bin
       FS params: size=98304, bs=4096, ps=128, es=4096
         Adding index.html: 70
         Adding ca.pem: 18826
         Adding conf0.json: 494
       FS stats : space total=84009, used=20418, free=63591
    CC    /vagrant/mos-test/stm32-test/build/gen/ffi_exports.c
    GEN   /vagrant/mos-test/stm32-test/build/gen/mg_build_info.c
    AR    /vagrant/mos-test/stm32-test/build/objs/mongoose-os.a
    GEN   /vagrant/mos-test/stm32-test/build/gen/fs_img.c
    GEN   /vagrant/mos-test/stm32-test/build/gen/build_info.c
    CC    /vagrant/mos-test/stm32-test/build/gen/fs_img.c
    CC    /vagrant/mos-test/stm32-test/build/gen/mg_build_info.c
    CC    /vagrant/mos-test/stm32-test/build/gen/build_info.c
    LD    /vagrant/mos-test/stm32-test/build/objs/stm32-test.elf (BL= LDS=stm32_no_ota.ld)
    /vagrant/mos-test/stm32-test/build/objs/mongoose-os.a(stm32_hal.c.o): In function `device_get_mac_address':
    /mongoose-os/fw/platforms/stm32/src/stm32_hal.c:61: undefined reference to `mg_hash_sha1_v'
    /vagrant/mos-test/stm32-test/build/objs/mgos_rpc.c.o: In function `acl_parse_cb':
    /vagrant/mos-test/stm32-test/deps/rpc-common/src/mgos_rpc.c:292: undefined reference to `mg_mk_str_n'
    /vagrant/mos-test/stm32-test/deps/rpc-common/src/mgos_rpc.c:292: undefined reference to `mg_match_prefix_n'
    /vagrant/mos-test/stm32-test/deps/rpc-common/src/mgos_rpc.c:294: undefined reference to `mg_mk_str_n'
    /vagrant/mos-test/stm32-test/build/objs/mgos_rpc.c.o: In function `mgos_rpc_req_prehandler':
    /vagrant/mos-test/stm32-test/deps/rpc-common/src/mgos_rpc.c:330: undefined reference to `mg_mk_str'
    /vagrant/mos-test/stm32-test/deps/rpc-common/src/mgos_rpc.c:337: undefined reference to `mg_mk_str'
    /vagrant/mos-test/stm32-test/deps/rpc-common/src/mgos_rpc.c:358: undefined reference to `mg_vcmp'
    /vagrant/mos-test/stm32-test/deps/rpc-common/src/mgos_rpc.c:408: undefined reference to `mg_time'
    /vagrant/mos-test/stm32-test/build/objs/mgos_rpc.c.o: In function `mgos_rpc_channel_ws_out_cfg_from_sys':
    /vagrant/mos-test/stm32-test/deps/rpc-common/src/mgos_rpc.c:515: undefined reference to `mg_mk_str'
    /vagrant/mos-test/stm32-test/deps/rpc-common/src/mgos_rpc.c:517: undefined reference to `mg_mk_str'
    /vagrant/mos-test/stm32-test/deps/rpc-common/src/mgos_rpc.c:518: undefined reference to `mg_mk_str'

The linking errors continue.

Prior to this excerpt, the reps (like roc-common and base mongoose-os c files successfully compile).  But in that listing, I do not see `mongoose-os\common\mg_str.c` and other C files which define methods reported here.

How do I add these common C files to the build so that the linker sees them properly?

Also, it looks like there is no STM32 definition for `mg_hash_sha1_v` -- the mongoose-os repo only has them defined for esp32, esp8266, & cc32xx platforms.

rojer, 2018-08-20:
i'll need full build command line you're using

> it looks like there is no STM32 definition for mg_hash_sha1_v

could very well be, STM32 support is still work in progress

osterwood, 2018-08-20:
Your documentation makes it sounds like STM32 is in place, not WIP.  The v2.4 [blog post](https://mongoose-os.com/blog/mongoose_os_v2-4/) is literally titled "**Mongoose OS v.2.4 is released - supports STM32**" and you have [recommended hardware](https://mongoose-os.com/docs/quickstart/devboards.md) like the other processors.

I have created a [sample repository](https://github.com/CapableRobot/mos-stm32-test) which reproduces the issue from your Empty Sample Project.  Strangely, the errors I get in this new repo are slightly different than the other repo -- the only difference between them is when the dependency checkout occurred (now versus 10 days ago).

Build output is now this:

```
  GEN   /vagrant/mos-stm32-test/build/gen/fs_img.c
  GEN   /vagrant/mos-stm32-test/build/gen/build_info.c
  GEN   /vagrant/mos-stm32-test/build/gen/mg_build_info.c
  AR    /vagrant/mos-stm32-test/build/objs/mongoose-os.a
  CC    /vagrant/mos-stm32-test/build/gen/fs_img.c
  CC    /vagrant/mos-stm32-test/build/gen/build_info.c
  CC    /vagrant/mos-stm32-test/build/gen/mg_build_info.c
  LD    /vagrant/mos-stm32-test/build/objs/mos-stm32-test.elf (BL= LDS=stm32_no_ota.ld)
/vagrant/mos-stm32-test/build/objs/mgos_rpc.c.o: In function `mgos_rpc_common_init':
/vagrant/mos-stm32-test/deps/rpc-common/src/mgos_rpc.c:474: undefined reference to `mgos_get_mgr'
/vagrant/mos-stm32-test/build/objs/mg_rpc.c.o: In function `mg_rpc_get_channel_info_internal_by_dst':
/vagrant/mos-stm32-test/deps/rpc-common/src/mg_rpc/mg_rpc.c:216: undefined reference to `mgos_get_mgr'
/vagrant/mos-stm32-test/build/objs/mongoose-os.a(mgos_sys_config.c.o): In function `mgos_sys_config_init':
/mongoose-os/fw/src/mgos_sys_config.c:281: undefined reference to `mgos_wdt_set_feed_on_poll'
/mongoose-os/fw/src/mgos_sys_config.c:284: undefined reference to `mgos_get_mgr'
/vagrant/mos-stm32-test/build/objs/mongoose-os.a(mgos_init.c.o): In function `mgos_init':
/mongoose-os/fw/src/mgos_init.c:36: undefined reference to `mgos_set_enable_min_heap_free_reporting'
/vagrant/mos-stm32-test/build/objs/mgos_deps_init.c.o:(.rodata.descrs+0xc): undefined reference to `mgos_mongoose_init'
/vagrant/mos-stm32-test/build/objs/mongoose-os.a(mgos_hal_freertos.c.o): In function `mgos_mg_poll_cb':
/mongoose-os/fw/src/mgos_hal_freertos.c:106: undefined reference to `mongoose_poll'
/mongoose-os/fw/src/mgos_hal_freertos.c:107: undefined reference to `mgos_get_mgr'
/vagrant/mos-stm32-test/build/objs/mongoose-os.a(mgos_uart.c.o): In function `mgos_uart_configure':
/mongoose-os/fw/src/mgos_uart.c:185: undefined reference to `mgos_add_poll_cb'
collect2: error: ld returned 1 exit status
```

cpq, 2018-08-20:
We are fixing the build as we speak - and let you know when done.

rojer, 2018-08-22:
it took a bit longer than expected, but this problem has been fixed in 2.5.1 (just released)

closed 2018-08-22 by rojer

aasick, 2019-12-02:
I'm using stm32f4 board(stm32f446re) 

For directory creation i run below command:

$ mos clone demo-c
demo-c: Does not exist, cloning from "https://github.com/mongoose-os-apps/demo-c"...
demo-c: Checking out 2.16.0...
demo-c: Done, hash dbab4a41af2433396395d9321101b744e8fd0251
---------------------------------------------------------------------------------------------------------------------
To build for stm32f4:

$ mos build --platform stm32f4
Connecting to https://mongoose.cloud, user test
Uploading sources (2718 bytes)
Handling lib "zz_boards"...
Handling lib "core"...
Handling lib "demo-bundle"...
zz_boards: Does not exist, cloning from "https://github.com/mongoose-os-libs/boards"...
core: Does not exist, cloning from "https://github.com/mongoose-os-libs/core"...
demo-bundle: Does not exist, cloning from "https://github.com/mongoose-os-libs/demo-bundle"...
zz_boards: Checking out 2.16.0...
zz_boards: Done, hash 897a76f9b7306751dea4596f38df4486548ae2da
zz_boards: Hash is updated:  -> 897a76f9b7306751dea4596f38df4486548ae2da
zz_boards: Prepared local dir: "/data/fwbuild-volumes/2.16.0/apps/demo-c/stm32f4/build_contexts/build_ctx_228567440/deps/boards"
demo-bundle: Checking out 2.16.0...
demo-bundle: Done, hash 1e22c49075afa299a9fb560db59b7d87875163ed
demo-bundle: Hash is updated:  -> 1e22c49075afa299a9fb560db59b7d87875163ed
demo-bundle: Prepared local dir: "/data/fwbuild-volumes/2.16.0/apps/demo-c/stm32f4/build_contexts/build_ctx_228567440/deps/demo-bundle"
core: Checking out 2.16.0...
core: Done, hash 1157c736d02af127b7aa24bfb79b93de1081636a
core: Hash is updated:  -> 1157c736d02af127b7aa24bfb79b93de1081636a
core: Prepared local dir: "/data/fwbuild-volumes/2.16.0/apps/demo-c/stm32f4/build_contexts/build_ctx_228567440/deps/core"
Handling lib "aws"...
Handling lib "provision"...
Handling lib "rpc-uart"...
Handling lib "dash"...
Handling lib "shadow"...
aws: Does not exist, cloning from "https://github.com/mongoose-os-libs/aws"...
provision: Does not exist, cloning from "https://github.com/mongoose-os-libs/provision"...
dash: Does not exist, cloning from "https://github.com/mongoose-os-libs/dash"...
rpc-uart: Does not exist, cloning from "https://github.com/mongoose-os-libs/rpc-uart"...
Handling lib "rpc-ws"...
Handling lib "ca-bundle"...
Handling lib "rpc-service-config"...
rpc-ws: Does not exist, cloning from "https://github.com/mongoose-os-libs/rpc-ws"...
rpc-service-config: Does not exist, cloning from "https://github.com/mongoose-os-libs/rpc-service-config"...
Handling lib "fstab"...
ca-bundle: Does not exist, cloning from "https://github.com/mongoose-os-libs/ca-bundle"...
Handling lib "http-server"...
shadow: Does not exist, cloning from "https://github.com/mongoose-os-libs/shadow"...
Handling lib "neopixel"...
fstab: Does not exist, cloning from "https://github.com/mongoose-os-libs/fstab"...
http-server: Does not exist, cloning from "https://github.com/mongoose-os-libs/http-server"...
Handling lib "sntp"...
Handling lib "rpc-service-fs"...
neopixel: Does not exist, cloning from "https://github.com/mongoose-os-libs/neopixel"...
rpc-service-fs: Does not exist, cloning from "https://github.com/mongoose-os-libs/rpc-service-fs"...
sntp: Does not exist, cloning from "https://github.com/mongoose-os-libs/sntp"...
Handling lib "rpc-service-gpio"...
rpc-service-gpio: Does not exist, cloning from "https://github.com/mongoose-os-libs/rpc-service-gpio"...
rpc-ws: Checking out 2.16.0...
provision: Checking out 2.16.0...
fstab: Checking out 2.16.0...
fstab: Done, hash 82cf05dda7e817a4aa4518003957c35fdfb178dc
provision: Done, hash 86b02414390c2237219cf1a3720c1e78b54d8e78
rpc-ws: Done, hash e8e507f4a91eb96a0c664fa33319b2e4f0ed89b2
fstab: Hash is updated:  -> 82cf05dda7e817a4aa4518003957c35fdfb178dc
fstab: Prepared local dir: "/data/fwbuild-volumes/2.16.0/apps/demo-c/stm32f4/build_contexts/build_ctx_228567440/deps/fstab"
rpc-ws: Hash is updated:  -> e8e507f4a91eb96a0c664fa33319b2e4f0ed89b2
rpc-ws: Prepared local dir: "/data/fwbuild-volumes/2.16.0/apps/demo-c/stm32f4/build_contexts/build_ctx_228567440/deps/rpc-ws"
[... 80 lines cut ...]
atca: Hash is updated:  -> a6601b8d479e5b5fcf8855289d4fd25f14d964da
atca: Prepared local dir: "/data/fwbuild-volumes/2.16.0/apps/demo-c/stm32f4/build_contexts/build_ctx_228567440/deps/atca"
Handling lib "i2c"...
i2c: Does not exist, cloning from "https://github.com/mongoose-os-libs/i2c"...
i2c: Checking out 2.16.0...
i2c: Done, hash 4bf5e876983b1767e50eab2bb4a71f4b225520e9
i2c: Hash is updated:  -> 4bf5e876983b1767e50eab2bb4a71f4b225520e9
i2c: Prepared local dir: "/data/fwbuild-volumes/2.16.0/apps/demo-c/stm32f4/build_contexts/build_ctx_228567440/deps/i2c"
The flag --repo is not given, going to use mongoose-os repository
mongoose-os: Does not exist, cloning from "https://github.com/cesanta/mongoose-os"...
mongoose-os: Checking out 2.16.0...
mongoose-os: Done, hash 9b779cd72d75c3d31d819c014646b5e606bbd725
Fetching libmbedtls-stm32f4.a (2.16.0) from https://github.com/mongoose-os-libs/mbedtls/releases/download/2.16.0/libmbedtls-stm32f4.a...
Error: /go/src/github.com/mongoose-os/mos/mos/manifest_parser/manifest_parser.go:353: neither sources nor prebuilt binary exists for the lib "mbedtls" (or, if a library doesn't have any code by design, its mos.yml shouldn't contain "sources"). Fetch error was: [failed to download libmbedtls-stm32f4.a: mongoose-os-libs/mbedtls: no asset libmbedtls-stm32f4.a found in release 2.16.0: file does not exist]
[... cut ...]
Error: /build/mos-4w_2Ho/mos-2.16.0+adcad3f~xenial0/go/src/github.com/mongoose-os/mos/mos/build_remote.go:326: build failed
/build/mos-4w_2Ho/mos-2.16.0+adcad3f~xenial0/go/src/github.com/mongoose-os/mos/mos/build.go:220: 
/build/mos-4w_2Ho/mos-2.16.0+adcad3f~xenial0/go/src/github.com/mongoose-os/mos/mos/build.go:163: 
/build/mos-4w_2Ho/mos-2.16.0+adcad3f~xenial0/go/src/github.com/mongoose-os/mos/mos/main.go:196: build failed

When i specify esp32 after --platform its build successfully, but stm32f4 i'm getting error
My board supporting Mongoose or Not?

nliviu, 2019-12-02:
AFAIK, to build for stm32 you need to add `--build-var BOARD=aaa` where `aaa` is one of the supported boards defined in the [boards](https://github.com/mongoose-os-libs/boards/blob/master/mos.yml) library.

aasick, 2019-12-02:
Hello nliviu,
I'm using STM32F446RE

In **boards/mos.yml** follow boards only available:
-stm32l475vg
-DISCO-F746NG
-NUCLEO-F746ZG
-stm32f205rg

mine is not available, can i create configuration for STM32F4 series boards, What are the steps needs to be included for create config for new board.

rojer, 2019-12-02:
you can, yes. unfortunately, the process is not documented at present, but you can start with definitions you see for other boards and work from there.
the other place to look is https://github.com/cesanta/mongoose-os/tree/master/platforms/stm32/mk which contains a bunch of device file definitions - you'll need to add one for your device

silviob72, 2019-12-03:
Hi, I'm interested to run mongoose os on stm32f4 family in specific on f429zi that is similar to f746zg. My problems should be the prebuilt library for mbedtls beacause when I try to build the app it returns the error: `neither sources nor prebuilt binary exists for the lib "mbedtls"`. How I can build that library for stm32f4?

rojer, 2019-12-03:
that is indeed a problem. for now, you should be able to use the Electron variant, which is built for stm32f2, like so:

```
- origin: https://github.com/mongoose-os-libs/mbedtls
  variant: stm32-Electron
```