cesanta/mongoose-os #445: AP to AP+STA; failed to write config.

andyjjones1, 2018-08-17:
When trying to transition from AP to AP+STA mode; I get the following error message.
`"mgos_wifi_dev_ap_set WiFi AP: Failed to set config"`

Here is the code that causes it:
https://gist.github.com/andyjjones1/11ffc788a5dbaa180523c23688b60492

When I call `wifi_setup_softap_setup()`, it first disables STA mode, then calls `wifi_setup_softap_setup()`, which then calls `wifi_softap_set_config_current()` (SDK call).

If I call this function very early after booting the ESP8266 (within the first 10 seconds), the SDK call `wifi_softap_set_config_current()` succeeds 100 times out of 100.  However if I wait longer than 30 seconds after boot, wifi_setup_softap_setup() will fail the first time I call it only; it will succeed on any subsequent calls.

Log of successful case:
https://gist.github.com/andyjjones1/8c52883a58d85f4c8c8d81c9e99352f6

Log of unsuccessful case: 
https://gist.github.com/andyjjones1/a93522042698f581f87c0818f7d86bbb

andyjjones1, 2018-08-17:
Another bit of information; whether or not I call `wifi_setup_sta_disable()` before trying to get into `AP+STA` mode makes no difference.  The exact same symptoms occur as described above.

DrBomb, 2018-08-17:
Looks like you will need to dive into mOS' source code to figure out what is going on.

While SDK calls are available and encouraged to use. Things like setting up the wifi are done in startup, you will be better off just changing the config and rebooting the device.

andyjjones1, 2018-08-17:
If I get `WiFi AP: Failed to set config` message,  `mgos_system_restart()` doesn't help me.  The new configuration data after reboot does not stick and STA mode is not enabled.

andyjjones1, 2018-08-17:
**mos-tool version:** v. 20180730-103227/master@c8ecb691.
**mongoose-os verison:** c6063cb57 (Jul 18 2018)
**core:** 80fc97171 (Aug 4 2018)
**wifi lib:** 35277c2cb (Aug 2 2018)

Code to reproduce:
https://gist.github.com/andyjjones1/11ffc788a5dbaa180523c23688b60492

DrBomb, 2018-08-17:
This is such a roundabout way of setting up wifi configs!

Have you tried using this function in C? https://github.com/cesanta/mongoose-os/blob/master/fw/include/mgos_sys_config.h#L86

Or the mJS equivalent? https://github.com/mongoose-os-libs/mjs/blob/master/fs/api_config.js#L60

If rebooting is not an issue to you, using these functions to set the wifi object to what you need, then rebooting should do the work.

andyjjones1, 2018-08-20:
Thanks Dr.Bomb,
I tried the mgos_config_apply() according to the way dimonomid used it his post [here](https://github.com/cesanta/mongoose-os/issues/365#issuecomment-351871868).  I have tried it 14 times so far (across timespans ranging from 30sec till 8min after boot) and it has not failed once yet.

The original method I was using to save the STA config was taken from [this page](https://forum.mongoose-os.com/discussion/1952/turn-off-ap-after-sta-connect-to-wifi) of the mongoose forum (starting a couple of posts in).

I guess the developers might chime in at some stage and tell us why this way is more stable?

Thanks again for your help.

andyjjones1, 2018-08-20:
I tried the `mgos_config_apply()` method according to the way dimonomid used it his post [here](https://github.com/cesanta/mongoose-os/issues/365#issuecomment-351871868). I have tried it 14 times so far (across timespans ranging from 30sec till 8min after boot) and it has not delivered the infamous "_failed to write config_" error message.

However, the actual SOFTAP mysoftAP won't show up until I press the button a second time, just as in the original post.  So in other words, the issue is the same as in the original post, just without the error message.

Thanks heaps for your help though!

DrBomb, 2018-08-20:
You will need to call a system reboot after changing the wifi configurations. That is normal.

andyjjones1, 2018-08-20:
Found a solution.  I inserted a 500 second delay between `wifi_setup_sta_disable()` (disabling of STA) and `wifi_setup_softap_setup()` (enabling of SOFTAP).  I have tried it 14 times in the same manner as my previous post, with the WiFi AP appearing each time to other devices in the room.

Code is here: https://gist.github.com/andyjjones1/a056439eba0486ba969e9cec0c0e9ae6

I'm not quite sure how, or why this works.  But someone may be interested in the future if they want to set up STA mode without rebooting.

rojer, 2018-08-20:
@andyjjones1 i tooka  quick look and can confirm you observation.
it seems that internally some transition happens some time after initial STA connection.
the trabsition from STA to AP goes fine if triggered before `pm open,type:2 0` message appears in the log (about 10 seconds after connection) and fails if it happens after.
judging by event in the log, it seems that before this transition wifi disconnection has immediate effect and afterwards it requires a task switch to take effect, which is what your timer workaround accomplishes. esp8266 is kind of a special snowflake, with no real rtos, which is part of the problem - the code has tor elinquish control before wifi stack gets to run again.

i thought about ways to fix this, and it's not trivial. given that we generally say that changing wifi configuration requires a reboot, i'm going to close this one without further action. i'm glad you found a workaround, and hopefully others will too, now that it's published here.

closed 2018-08-20 by rojer