cesanta/mongoose-os #448: stm32 build error

hoseung2, 2018-08-31:
Using STM32 board, I want to build `c_hello` example files. But, there are some errors like theses.

I use this cmd: `mos build --arch=stm32 mos.yml`

`Error: neither sources nor prebuilt binary exists for the lib "mongoose" (or, if a library doesn't have any code by design, its mos.yml shouldn't contain "sources"). Fetch error was: [got 404 status code when accessed https://github.com/mongoose-os-libs/mongoose/releases/download/2.5.1/libmongoose-stm32.a]
Error: build failed`

So, I changed the name of the architecture to that specified in the release files and the following error occurred, saying that it does not support the specified platform.

`mos build --arch=stm32-DISCO-F746NG mos.yml`

`Successfully fetched prebuilt binary for "mongoose" to "/home/khs/mongoose-os/fw/examples/c_hello/deps/mongoose/lib/stm32-disco-f746ng/libmongoose.a"
Successfully fetched prebuilt binary for "ota-http-client" to "/home/khs/mongoose-os/fw/examples/c_hello/deps/ota-http-client/lib/stm32-disco-f746ng/libota-http-client.a"
Successfully fetched prebuilt binary for "ota-http-server" to "/home/khs/mongoose-os/fw/examples/c_hello/deps/ota-http-server/lib/stm32-disco-f746ng/libota-http-server.a"
Error: can't build for the platform stm32-disco-f746ng; only those platforms are supported: [cc3200 esp32 esp8266 stm32]`

What should I do??

nliviu, 2018-08-31:
Add `--build-var BOARD:DISCO-F746NG` to the build command, or add
```
build_vars:
  BOARD: DISCO-F746NG
```
in `mos.yml`
It built ok for me.

PS. I don't have a STM32 board. My command line was `mos build --local --platform stm32 --build-var BOARD:DISCO-F746NG` using `mos` latest.

rojer, 2018-09-17:
yes, for STM32 you need additional specifications - at least device family (F4, L7 or L4). also you need prebuilt binaries, which we currently prbuild only for F7 and L4. the upshot is, that full support is only available for 3 boards: B-L475E-IOT01A, DISCO-F746NG and NUCLEO-F746ZG

closed 2018-09-17 by rojer