cesanta/mongoose-os #450: uart buffer overflows and out of sync buffer/available bytes

rtrue983, 2018-09-17:
I'm experiencing a few problems with the UART that i tried to explain above but let me try to summarize again. I'm a bit at a loss as to what is causing it but this is what I'm seeing.

1. If data comes in too quickly, the EPS32 / mgos uart buffer will keep data (7 to 20 bytes of data) in its buffer and indicate that 0 bytes are available. I have confirmed that I have no loss of data and that my message parser can handle the breaks, but if data is being streamed in regularly it can leave large gaps in real time messaging. 

2. I also see rx_overflows. This almost always causes problem 1 to occur, but problem 1 can occur without an overflow happening. I can't confirm this yet but I think it is caused by sending data while also trying to receive. and or pulling the bytes out of the rx buffer 1 at a time while data is coming in and data is going out. I would like to keep things asynchronous but can't seem to get things to play happy at the moment.

3. Once a problem occurs it seems like the uart driver can never fully recover. (this is where i would like to reset the uart driver)

4. I still don't understand the idea of the dispatcher callback... I see the two spots where the uart dispatcher is called (tx_av < threshold and rx_ava > 0) but what I'm seeing is that mongoose_call is always calling it if if the conditions are not met. yes i can filter on what i need to but it seems more efficient just to give a interrupt callback when it occurs instead of just polling the variables.

Since then, I'm in the works of making thins a little more synchronous but am still experiencing overflow issues so i'm a little puzzled as to what I'm doing wrong.  

Screenshot of messages
[image]
The app detected there was an overflow and prints the status at 5.5sec (blue)... so from startup at 2.7sec to then I occurred a buffer overflow.  
Things to note, the master is sending out a heartbeat every 20ms, if the wifi needs to send a command it does so after a heartbeat is received.  Commands can also be sent from the main, if they are, a 100ms space is left for a response to occur

*Zoomed In (start)*
[image]
2:85ms wifi powers up
notice the heartbeat is coming in before and after the startup... not sure if that matters

*Zoomed In (first message handshake)*
[image]
3s:63ms heartbeat (from main) - yellow
3s:70ms command (to main) - green
3s:83ms response ACK (from main) - yellow

*LOG Output*
``` 
[Sep 17 13:35:23.298] mgos_init            Init done, RAM: 279264 total, 132880 free, 132868 min free
[Sep 17 13:35:23.304] mgos_aws_shadow_init AWS Device Shadow requires MQTT
[Sep 17 13:35:23.309] mongoose_poll        New heap free LWM: 132772
[Sep 17 13:35:23.315] mgos_net_on_change_c WiFi STA: connecting
[Sep 17 13:35:23.322] wifi_cb              MGOS_NET_EV_CONNECTING staState=0
[Sep 17 13:35:23.329] wifi_cb              WiFi EVT: Connecting (staState = STA_CONNECTING)
[Sep 17 13:35:23.334] GPIO_StartLedFlash   LED1 Start Flash: 100 200 1400
[Sep 17 13:35:23.351] GPIO_StartLedFlash   LED0 Start Flash: 100 200 1400
[Sep 17 13:35:23.356] mongoose_poll        New heap free LWM: 132324
[Sep 17 13:35:23.361] mongoose_poll        New heap free LWM: 132224
[Sep 17 13:35:23.367] mongoose_poll        New heap free LWM: 132044
[Sep 17 13:35:23.376] mongoose_poll        New heap free LWM: 130628
[Sep 17 13:35:23.381] mongoose_poll        New heap free LWM: 130528
[Sep 17 13:35:23.939] I (2676) wifi: n:6 1, o:6 0, ap:6 1, sta:6 0, prof:6
[Sep 17 13:35:24.791] I (3526) wifi: state: init -> auth (b0)
[Sep 17 13:35:24.802] UartRxTask           uart rx overflows occured [rx_ovf:2]
[Sep 17 13:35:24.808] mongoose_poll        New heap free LWM: 118656
[Sep 17 13:35:24.819] UartRxTask           uart stats | app mIdx:  0 av:  0 bytes:280 | rx i:9 b:280 of:2 lc:90 | tx i:2 b:14 th:0
[Sep 17 13:35:25.795] I (4536) wifi: state: auth -> init (2)
[Sep 17 13:35:25.800] I (4536) wifi: n:6 0, o:6 1, ap:6 1, sta:6 0, prof:6
```

---

**UART Init Settings**
``` c
    mgos_uart_config_set_defaults(APP_UART_NO, &ucfg);

    ucfg.baud_rate = UART_BAUD_RATE;
    ucfg.num_data_bits = 8;
    ucfg.parity = MGOS_UART_PARITY_NONE;
    ucfg.stop_bits = MGOS_UART_STOP_BITS_1;
    ucfg.rx_fc_type = MGOS_UART_FC_NONE;
    ucfg.rx_linger_micros = 150;
    ucfg.rx_buf_size = 1500;
    ucfg.tx_buf_size = 1500;

    if (!mgos_uart_configure(APP_UART_NO, &ucfg)) {
        LOG(LL_INFO, ("ERROR: failed to init uart"));
  	    return false; // failed to configure uart
[... 163 lines cut ...]
{
    static uartMsg_t uartMsg;
    uint16_t mLen = pLen+MSG_OVERHEAD;

    if (mLen > 1048) {
        LOG( LL_ERROR, (PR "msg len too long" PW ) );
        return 0;
    }

    // [BUILD MESSAGE]

    if (pLen) {
        memcpy( &uartMsg.data[PAYLOAD_START_BYTE], payload, pLen );
    }
    // [CALCULATE CRC]
    
    uartMsg.type = RESPONSE;
    uartMsg.len = mLen;

    // send the data out
    txCount += mgos_uart_write( APP_UART_NO, uartMsg.data, uartMsg.len );

    return mLen;
}
```

rtrue983, 2018-09-19:
A little update.  I have performed the following things, each increasing the reliablity:

1. main micro controlling the flow as the master.  I believe this helps prevent any receiving while transmitting on the ESP32... the rx_overflows/buffer offsets occur less frequent.  

2. Processed the incoming commands immidately instead of sending them to a queue.  Prevents mongoose from getting busy doing other things.  **QUESTION:** how long can i hold up the rx dispatcher callback... I assume it is called from the mgos task and not an interrupt? 

3. added a `mgos_uart_flush( ... )` after I call `mgos_uart_write( ... )` for sending the commands back. 
 I have to do some more testing on this but I think this cleans things up quite significantly.  I don't really like solution cause if we have a really busy uart we will be blocked quite a bit right?  Right now the only thing calling the send msg/flush is the uart rx dispatcher.

**UPDATED SOURCE CODE**

*initialization*
``` c
bool UART_Init( void )
{
    struct mgos_uart_config ucfg;

    if (isInitialized == false) {

        mgos_uart_config_set_defaults(APP_UART_NO, &ucfg);

        ucfg.baud_rate = UART_BAUD_RATE;
        ucfg.num_data_bits = 8;
        ucfg.parity = MGOS_UART_PARITY_NONE;
        ucfg.stop_bits = MGOS_UART_STOP_BITS_1;
        ucfg.rx_fc_type = MGOS_UART_FC_NONE;
        ucfg.rx_linger_micros = 15;
        ucfg.rx_buf_size = 1500;
        ucfg.tx_buf_size = 1500;

        if (!mgos_uart_configure(APP_UART_NO, &ucfg)) {
            LOG(LL_INFO, ("ERROR: failed to init uart"));
            return false; // failed to configure uart
        }

        uartRxQueue =    xQueueCreate( 10, sizeof(uartMsg_t) );

        xTaskCreate( UartRxTask,    "uartRxTask",    8192, NULL, APP_HIGH_TASK_PRIORITY, NULL);

        // setup the receive callback and enable the receiver
        mgos_uart_set_dispatcher( APP_UART_NO, UartRxCB, NULL );
        mgos_uart_set_rx_enabled( APP_UART_NO, true );

        UrtDrv_EnableDataPassThru( false, 0);
        isInitialized = true;
    }

    return true; // successfully initialized the uart peripheral
}
```

*rx dispatcher*
```` c
/// @brief    This private function is a mongoose callback gets called from 
///             their event task when bytes are available in their receive buffer
///             or if there is room in their output buffer (so pretty much all
///             the time)
///
/// @param    [in] uart_no
/// @param    [in/out] *arg
/// @returns  None.
static void UartRxCB( int uart_no, void *arg )
{
    uint8_t rxData = 0;
    size_t rx_av = 0;
    uartMsg_t uartRxMsg;

    if (uart_no != APP_UART_NO) {
        return;
    }
[... 98 lines cut ...]
}
```

*transmit command/heartbeat response*
``` c 
void UrtDrv_SendNow( uint8_t count )
{
    static uartTxMsg_t uartTxMsg;

    // is there a message to send? if so, send it now
    if (pdTRUE == xQueueReceive( uartTxQueue, &uartTxMsg, 0 )) {
        
        // send the data out
        txCount += mgos_uart_write( APP_UART_NO, uartTxMsg.data, uartTxMsg.len );
    }
    else {

        // ASSEMBLE RESPONSE TO HEARTBEAT

        txCount += mgos_uart_write( APP_UART_NO, txMsg, MSG_OVERHEAD+1 );
    }

    mgos_uart_flush(APP_UART_NO);
}
```

rtrue983, 2018-09-19:
Here is a good scope grab of what i'm seeing once in error state (rx_av = 0 but data is stuck in the lower driver):

**Zoomed Out**
main micro transmits messages (yellow) every 50-60ms.  
esp32 responds once message is received (green)
[image]

**Zoomed In**
notice the command is processed and the response is sent after 1 byte was received (purple toggles on every `mgos_uart_read( ..., 1 )`).  This indicates the first 7 bytes of the message were received and handled previously and then 0 bytes were available until this message came in.
notice the purple toggling after the response... these are the first 7 bytes available of the message shown
[image]

ouraborous, 2018-12-20:
Hello,

I don't know if you are still facing this problem. But for the application I am working on, I was having the same issue. Not sure yet if the trigger is high activity or concurrent usage of multiple serial ports. But I can confirm that there is a HW issue on the ESP32 that causes the Rx Fifo and the Rx Count register to get out of sync. As you mentioned, the rx?count is returning 0 when there are bytes available, and after more bytes are received the older bytes are sent with a portion of the new ones and the overall message is corrupted. I know Espressif have accepted that there is an issue with the RXFIFO_RD_BYTE register and recommends to access it via AHB and Mongoose implements this in the rx_byte function on exp32_uart.c

But i have found open issues with this same out of sync problem between the rx_cnt and FIFO. And ultimately the solution exposed in [this](https://github.com/espressif/arduino-esp32/pull/1849) thread makes sense. I implemented this solution in the esp32_uart.c driver and I have not seen the issue occur again. Basically what this does is to use the write/read addresses of the Rx FIFO to know if it´s not empty instead of reading the RX Count register. Is less efficient in terms of memory management but much more reliable. Let me know it you already tested something like this and if it works for you. Maybe we can get a fix in the mongoose OS code. I implemented this function:

```
IRAM static bool esp32_uart_rx_fifo_not_empty(int uart_no) {
  int rx_add = REG_GET_FIELD(UART_MEM_RX_STATUS_REG(uart_no), UART_MEM_RX_RD_ADDR);
  int tx_add = REG_GET_FIELD(UART_MEM_RX_STATUS_REG(uart_no), UART_MEM_RX_WR_ADDR);
  return (rx_add != tx_add);
}
```

and modified this one in esp32_uart.c

```
void mgos_uart_hal_dispatch_rx_top(struct mgos_uart_state *us) {
  int uart_no = us->uart_no;
  struct mbuf *rxb = &us->rx_buf;
  uint32_t rxn = 0;
  /* RX */
  if (mgos_uart_rxb_free(us) > 0 && esp32_uart_rx_fifo_not_empty(uart_no)) {
    int linger_counter = 0;
    /* 32 here is a constant measured (using system_get_time) to provide
     * linger time of rx_linger_micros. It basically means that one iteration
     * of the loop takes 3.2 us.
     *
     * Note: lingering may starve TX FIFO if the flow is bidirectional.
     * TODO(rojer): keep transmitting from tx_buf while lingering.
     */
    int max_linger = us->cfg.rx_linger_micros / 10 * 32;
    while (mgos_uart_rxb_free(us) > 0 && linger_counter <= max_linger) {
      if (esp32_uart_rx_fifo_not_empty(uart_no)) {
        mbuf_resize(rxb, rxb->len + 1);
        uint8_t b = rx_byte(uart_no);
        mbuf_append(rxb, &b, 1);
        rxn++;
        if (linger_counter > 0) {
          us->stats.rx_linger_conts++;
          linger_counter = 0;
        }
      } else {
        linger_counter++;
      }
    }
    us->stats.rx_bytes += rxn;
  }
  WRITE_PERI_REG(UART_INT_CLR_REG(uart_no), UART_RX_INTS);
}
```

rojer, 2019-01-11:
i believe this was finally fixed by https://github.com/cesanta/mongoose-os/commit/34aa64d86baefa4ce167c0ab8c0cf3f556b44011

closed 2019-01-11 by rojer