cesanta/mongoose-os #454: Authenticated RPC replay attack vulnerability over BLE.

JamesHepworth, 2018-09-19:
Good morning,

Our application calls RPCs over BLE using the rpc-gatts lib as well as over websockets. We also require our RPCs to be secured and and have implemented authentication and authorisation as descirbed [here](https://mongoose-os.com/docs/userguide/security.md). 

On testing it appears that Mongoose OS has a potential vulnerability to replay attacks over BLE when rpc frames are recorded over websockets or COM ports.

With the device plugged into a comm port with debug level set to 3, I performed a secure RPC pver websockets with the correct authentication details and logged the frame using Wireshark as follows:

> {"src":"mos","id":1899842071503,"method":"Sys.GetInfo","auth":{"realm":"esp32_Test_Board","username":"James","nonce":1537347057,"cnonce":867183336,"response":"f421216246c2e3907036b0c81e9cbe4c"}}

Associated log entries are:

> Sep 19 10:50:54.985] mg_rpc_ev_handler    0x3ffd3468 GOT FRAME (195): {"src":"mos","id":1899842071503,"method":"Sys.GetInfo","auth":{"realm":"esp32_Test_Board","username":"James","nonce":1537347057,"cnonce":867183336,"response":"f421216246c2e3907036b0c81e9cbe4c"}}
[Sep 19 10:50:54.993] 
[Sep 19 10:50:54.993] mg_rpc_parse_frame   1899842071503 'mos' '' 'Sys.GetInfo'
[Sep 19 10:50:55.001] find_mount_by_path   rpc_acl.json -> /rpc_acl.json pl 1 -> 1 0x3ffdb900
[Sep 19 10:50:55.009] mgos_vfs_open        rpc_acl.json 0x0 0x1b6 => 0x3ffdb900 rpc_acl.json 1 => 257 (refs 1)
[Sep 19 10:50:55.014] mgos_vfs_fstat       257 => 0x3ffdb900:1 => 0 (size 939)
[Sep 19 10:50:55.020] mgos_vfs_fstat       257 => 0x3ffdb900:1 => 0 (size 939)
[Sep 19 10:50:55.024] mgos_vfs_lseek       257 0 1 => 0x3ffdb900:1 => 0
[Sep 19 10:50:55.029] mgos_vfs_lseek       257 0 0 => 0x3ffdb900:1 => 0
[Sep 19 10:50:55.034] mgos_vfs_lseek       257 0 0 => 0x3ffdb900:1 => 0
[Sep 19 10:50:55.039] mgos_vfs_lseek       257 0 2 => 0x3ffdb900:1 => 939
[Sep 19 10:50:55.044] mgos_vfs_lseek       257 0 0 => 0x3ffdb900:1 => 0
[Sep 19 10:50:55.055] mgos_vfs_close       257 => 0x3ffdb900:1 => 0 (refs 0)
[Sep 19 10:50:55.064] mgos_rpc_req_prehand Called 'Sys.GetInfo', acl for it: '+James'
[Sep 19 10:50:55.077] mg_rpc_check_digest_ Got auth: Realm:esp32_Test_Board, Username:James, Nonce: 1537347057, CNonce:867183336, Response:f421216246c2e3907036b0c81e9cbe4c
[Sep 19 10:50:55.084] find_mount_by_path   rpc_auth.txt -> /rpc_auth.txt pl 1 -> 1 0x3ffdb900
[Sep 19 10:50:55.096] mgos_vfs_open        rpc_auth.txt 0x0 0x1b6 => 0x3ffdb900 rpc_auth.txt 1 => 257 (refs 1)
[Sep 19 10:50:55.102] mgos_vfs_fstat       257 => 0x3ffdb900:1 => 0 (size 56)
[Sep 19 10:50:55.112] mg_check_digest_auth James esp32_Test_Board f421216246c2e3907036b0c81e9cbe4c f421216246c2e3907036b0c81e9cbe4c
[Sep 19 10:50:55.118] mgos_vfs_close       257 => 0x3ffdb900:1 => 0 (refs 0)
[Sep 19 10:50:55.120] mg_rpc_check_digest_ Auth ok
[Sep 19 10:50:55.126] mg_rpc_handle_reques Sys.GetInfo via WS_in 192.168.1.157:57627
[Sep 19 10:50:55.133] find_mount_by_path   / -> / pl 1 -> 1 0x3ffdb900
[Sep 19 10:50:55.138] find_mount_by_path   / -> / pl 1 -> 1 0x3ffdb900
[Sep 19 10:50:55.161] mg_rpc_send_frame    0x3ffd3468 SEND FRAME (390): {"id":1899842071503,"src":"device","dst":"mos","result":{"app": "app", "fw_version": "1.0", "fw_id": "20180919-072723", "mac": "xxxx", "arch": "esp32", "uptime": 124, "ram_size": 217424, "ram_free": 112912, "ram_min_free": 101784, "fs_size": 233681, "fs_free": 207577,"wifi": {"sta_ip": "xxxx", "ap_ip": "xxxx", "status": "got ip", "ssid": "xxxx"}}} -> 1

Using a BLE client I then passed the following frame to the device as a series of messages:

> {"method":"Sys.GetInfo","auth":{"realm":"esp32_Test_Board","username":"James","nonce":1537347057,"cnonce":867183336,"response":"f421216246c2e3907036b0c81e9cbe4c"}}}

This frame is a copy of the earlier frame recorded without the "src" and "id" keys. The associated logs are as follows:

> [Sep 19 10:57:00.304] mg_rpc_ev_handler    0x3ffd394c GOT FRAME (164): {"method":"Sys.GetInfo","auth":{"realm":"esp32_Test_Board","username":"James","nonce":1537347057,"cnonce":867183336,"response":"f421216246c2e3907036b0c81e9cbe4c"}}}
[Sep 19 10:57:00.309] mg_rpc_parse_frame   0 '' '' 'Sys.GetInfo'
[Sep 19 10:57:00.317] find_mount_by_path   rpc_acl.json -> /rpc_acl.json pl 1 -> 1 0x3ffdb900
[Sep 19 10:57:00.326] mgos_vfs_open        rpc_acl.json 0x0 0x1b6 => 0x3ffdb900 rpc_acl.json 1 => 257 (refs 1)
[Sep 19 10:57:00.331] mgos_vfs_fstat       257 => 0x3ffdb900:1 => 0 (size 939)
[Sep 19 10:57:00.337] mgos_vfs_fstat       257 => 0x3ffdb900:1 => 0 (size 939)
[Sep 19 10:57:00.341] mgos_vfs_lseek       257 0 1 => 0x3ffdb900:1 => 0
[Sep 19 10:57:00.346] mgos_vfs_lseek       257 0 0 => 0x3ffdb900:1 => 0
[Sep 19 10:57:00.351] mgos_vfs_lseek       257 0 0 => 0x3ffdb900:1 => 0
[Sep 19 10:57:00.356] mgos_vfs_lseek       257 0 2 => 0x3ffdb900:1 => 939
[Sep 19 10:57:00.360] mgos_vfs_lseek       257 0 0 => 0x3ffdb900:1 => 0
[Sep 19 10:57:00.368] mgos_vfs_close       257 => 0x3ffdb900:1 => 0 (refs 0)
[Sep 19 10:57:00.376] mgos_rpc_req_prehand Called 'Sys.GetInfo', acl for it: '+James'
[Sep 19 10:57:00.389] mg_rpc_check_digest_ Got auth: Realm:esp32_Test_Board, Username:James, Nonce: 1537347057, CNonce:867183336, Response:f421216246c2e3907036b0c81e9cbe4c
[Sep 19 10:57:00.396] find_mount_by_path   rpc_auth.txt -> /rpc_auth.txt pl 1 -> 1 0x3ffdb900
[Sep 19 10:57:00.409] mgos_vfs_open        rpc_auth.txt 0x0 0x1b6 => 0x3ffdb900 rpc_auth.txt 1 => 257 (refs 1)
[Sep 19 10:57:00.414] mgos_vfs_fstat       257 => 0x3ffdb900:1 => 0 (size 56)
[Sep 19 10:57:00.425] mg_check_digest_auth James esp32_Test_Board f421216246c2e3907036b0c81e9cbe4c f421216246c2e3907036b0c81e9cbe4c
[Sep 19 10:57:00.430] mgos_vfs_close       257 => 0x3ffdb900:1 => 0 (refs 0)
[Sep 19 10:57:00.433] mg_rpc_check_digest_ Auth ok
[Sep 19 10:57:00.440] mg_rpc_handle_reques Sys.GetInfo via GATTS if 4 conn 0 peer d4:6d:6d:a5:22:a5,0
[Sep 19 10:57:00.447] find_mount_by_path   / -> / pl 1 -> 1 0x3ffdb900
[Sep 19 10:57:00.452] find_mount_by_path   / -> / pl 1 -> 1 0x3ffdb900
[Sep 19 10:57:00.489] mg_rpc_send_frame    0x3ffd394c SEND FRAME (359): {"src":"device","result":{"app": "app", "fw_version": "1.0", "fw_id": "20180919-072723", "mac": "xxxx", "arch": "esp32", "uptime": 489, "ram_size": 217332, "ram_free": 113068, "ram_min_free": 101784, "fs_size": 233681, "fs_free": 207577,"wifi": {"sta_ip": "xxxx", "ap_ip": "xxxx", "status": "got ip", "ssid": "xxxx"}}} -> 1

Clearly the second request was also authorised and the RPC called by simply repassing the recorded authenticated frame to the device. Anyone who is able to monitor the AP traffic and record the frames with realm, username, nonce, cnonce and response keys is potentially able to call an RPC on the device without valid authentication.

This behaviour was also evident when the initial, authenticated rpc was called and recorded over a COM  port.

Attached are the complete device logs for the test.
[Device logs.txt](https://github.com/cesanta/mongoose-os/files/2396506/Device.logs.txt)

Thanks,
James

duncandrennan, 2018-09-19:
To add to this, I think the underlying issue is that the nonce counter on the server (device) side is not handled correctly.

In rpc_common in mg_rpc.c line 864 the following is called

```
int authenticated = mg_check_digest_auth(
            mg_mk_str("dummy_method"), mg_mk_str("dummy_uri"), username, cnonce,
            response, mg_mk_str("auth"), mg_mk_str("1"), nonce, realm,
            htdigest_fp);
```

with

```
int mg_check_digest_auth(struct mg_str method, struct mg_str uri,
                         struct mg_str username, struct mg_str cnonce,
                         struct mg_str response, struct mg_str qop,
                         struct mg_str nc, struct mg_str nonce,
                         struct mg_str auth_domain, FILE *fp)
```

so nc is a fixed value of '1'. The server should increment nc (or prevent previous nc values being reused for a particular nonce/cnone) after each authentication so that cnonce and nonce can be reused to prevent having to exchange authentication info for each transaction.

If nc is a fixed value it allows a replay attack - see nonce-count in https://tools.ietf.org/html/rfc2617

rojer, 2018-10-01:
ok, i will close it as "not working as not intended" then.

closed 2018-10-01 by rojer