cesanta/mongoose-os #458: Flash Errors

gopalakrishnan-subramani, 2018-10-13:
Getting this error every time when I try to flash. I connected to ESP32, flash is working fine only after 3-5 retries, never succeed on first attempt.  Using the ESP32 USB connecting, running on Mac machine, anyone give what is wrong, how to succeed flash on first attempt.

$ mos build --arch esp32 && mos flash
Connecting to https://mongoose.cloud, user test
Uploading sources (7783 bytes)
Firmware saved to build/fw.zip
Loaded esp-sense/esp32 version 1.0 (20181013-125738)
Using port /dev/cu.SLAB_USBtoUART
Opening /dev/cu.SLAB_USBtoUART @ 115200...
Connecting to ESP32 ROM, attempt 1 of 10...
Connecting to ESP32 ROM, attempt 2 of 10...
Connecting to ESP32 ROM, attempt 3 of 10...
  Connected, chip: ESP32D0WDQ6 R1
Running flasher @ 0...
  Flasher is running
Error: /private/tmp/mos-20181005-44753-1lei2ig/mos-tool-b997b0fde26931ac88a11a1fc6276c52c36187c8/src/cesanta.com/mos/flash/common/slip.go:38: invalid SLIP starting byte: 0xc8
/private/tmp/mos-20181005-44753-1lei2ig/mos-tool-b997b0fde26931ac88a11a1fc6276c52c36187c8/src/cesanta.com/mos/flash/esp/flasher/flasher_client.go:128: error reading response packet
/private/tmp/mos-20181005-44753-1lei2ig/mos-tool-b997b0fde26931ac88a11a1fc6276c52c36187c8/src/cesanta.com/mos/flash/esp/flasher/flasher_client.go:165:
/private/tmp/mos-20181005-44753-1lei2ig/mos-tool-b997b0fde26931ac88a11a1fc6276c52c36187c8/src/cesanta.com/mos/flash/esp/flasher/common.go:94: failed to get flash chip id
/private/tmp/mos-20181005-44753-1lei2ig/mos-tool-b997b0fde26931ac88a11a1fc6276c52c36187c8/src/cesanta.com/mos/flash/esp/flasher/common.go:70: flash size is not specified and could not be detected
/private/tmp/mos-20181005-44753-1lei2ig/mos-tool-b997b0fde26931ac88a11a1fc6276c52c36187c8/src/cesanta.com/mos/flash/esp/flasher/flash.go:44:
/private/tmp/mos-20181005-44753-1lei2ig/mos-tool-b997b0fde26931ac88a11a1fc6276c52c36187c8/src/cesanta.com/mos/flash.go:157:
/private/tmp/mos-20181005-44753-1lei2ig/mos-tool-b997b0fde26931ac88a11a1fc6276c52c36187c8/src/cesanta.com/mos/main.go:151:
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

Rosmianto, 2018-10-31:
Have you tried with another ESP32 module? It seems that you experiencing USB connection issue.

gopalakrishnan-subramani, 2018-11-03:
I  found that using this command 
```
 mos build --arch esp32 && mos flash
```

fails while uploading firmware. 

But when I do them separately , flash is working fine. 

``` 
mos build --arch esp32

 mos flash
```

closed 2018-11-03 by gopalakrishnan-subramani

aasick, 2019-12-02:
for me when i closed serial terminal, at the time only file loading successfully : )

After build connect board with host machine(Don't open serial console).
Now you run $mos flash
After flash completed you open serial console through putty or minicom

; )