cesanta/mongoose-os #462: ESP8266 - PWM limit at 10kHz

pankleks, 2018-11-03:
Hello,

It looks like PWM freq. is limited to 10kHz directly in the code:

https://github.com/mongoose-os-libs/pwm/blob/master/src/esp8266/esp_pwm.c#L150

10kHz is not enough for applications like FAN control.

ESP8266 is capable of running PMW at 25kHz - tested with direct Arduino code.

Forum discussion for ref:

https://forum.mongoose-os.com/discussion/3735/pwm-at-25khz

Thanks!

pimvanpelt, 2018-11-04:
If you're controlling fans, I recommend you use a [PWM Fan Controller](https://www.microchip.com/wwwproducts/en/EMC2305). I have this particular unit in a deployment, and it works just fine. You can set PWM speeds (both RPM and duty cycle) via I2C.
That sounds a lot more reliable than using bitbanging PWM from the MCU (and as a bonus, the chip will give you 5 channels).

pankleks, 2018-11-04:
Thanks - looks like neat chip, although it adds another component to the system and if 8266 is capable of doing something I find no reason to add separate component.

I'm currently running 8266 controlling 2 fans independently on separate pins (running arduino code) for 24/7 - all is stable and chip is not even hot.

Chip you mentioned is cheap but it still cost nearly same as 8266 itself:)

rojer, 2018-11-07:
ok, in my testing 25 KHz definitely works, even 40 KHz does.
i'm removing the check and let you guys figure out what makes sense for yourself.

closed 2018-11-07 by rojer