cesanta/mongoose-os #472: mos build failing for esp32

loui4u, 2019-01-03:
Hi,

I have cloned demo-js application and tried to build firmware. I am getting error with mbedtls library. Here is the error message. It was working couple of days ago.
```
The flag --repo is not given, going to use mongoose-os repository
mongoose-os: Checking out master...
mongoose-os: Pulling...
mongoose-os: Done, hash 31f9fe69869f851fa2395ec62d4547bed3819226
Fetching libmbedtls-esp32.a (latest) from https://github.com/mongoose-os-libs/mbedtls/releases/download/latest/libmbedtls-esp32.a...
E0103 04:42:10.765655       1 swmodule.go:349] GitHub asset libmbedtls-esp32.a download failed (attempt 1): mongoose-os-libs/mbedtls: no asset libmbedtls-esp32.a found in release latest
Fetching libmbedtls-esp32.a (latest) from https://github.com/mongoose-os-libs/mbedtls/releases/download/latest/libmbedtls-esp32.a...
E0103 04:42:12.069436       1 swmodule.go:349] GitHub asset libmbedtls-esp32.a download failed (attempt 2): mongoose-os-libs/mbedtls: no asset libmbedtls-esp32.a found in release latest
Fetching libmbedtls-esp32.a (latest) from https://github.com/mongoose-os-libs/mbedtls/releases/download/latest/libmbedtls-esp32.a...
E0103 04:42:13.362792       1 swmodule.go:349] GitHub asset libmbedtls-esp32.a download failed (attempt 3): mongoose-os-libs/mbedtls: no asset libmbedtls-esp32.a found in release latest
Error: /go/src/cesanta.com/mos/manifest_parser/manifest_parser.go:312: neither sources nor prebuilt binary exists for the lib "mbedtls" (or, if a library doesn't have any code by design, its mos.yml shouldn't contain "sources"). Fetch error was: [failed to download libmbedtls-esp32.a: mongoose-os-libs/mbedtls: no asset libmbedtls-esp32.a found in release latest]
/go/src/cesanta.com/mos/build_local.go:116:
/go/src/cesanta.com/mos/build.go:201:
/go/src/cesanta.com/mos/build.go:144:
/go/src/cesanta.com/mos/main.go:176: build failed
Error: /Users/mos/tmp/mos_gopath/src/cesanta.com/mos/build_remote.go:309: build failed
/Users/mos/tmp/mos_gopath/src/cesanta.com/mos/build.go:201:
/Users/mos/tmp/mos_gopath/src/cesanta.com/mos/build.go:144:
/Users/mos/tmp/mos_gopath/src/cesanta.com/mos/main.go:176: build failed
```

Thanks and appreciate your support.
Loui

loui4u, 2019-01-03:
Issue seems to be specific to my mac. It is working fine on ubuntu. Any help/direction is appreciated.  I am closing this case.

Loui

closed 2019-01-03 by loui4u

tcpipchip, 2019-01-21:
I am having the same problem on WINDOWS 7

dniadzelka, 2019-04-30:
have the same on mac

dniadzelka, 2019-05-01:
Duplicate of #500