cesanta/mongoose-os #486: Some phones are not issued IP and unable to connect due to IDF 3.2 bug

tripflex, 2019-03-08:
https://github.com/espressif/esp-idf/issues/2915

This is a deal breaker for me as unable to get any IP assigned from device --- only figured this out while trying to test my captive portal lib -- but now makes sense when my clients were telling me they were unable to connect and were not getting an IP on their phones -- which is a HUGE issue for me as this renders the device basically useless for local setup and configuration 

Specifically this comment says it has not been pushed to 3.2:
https://github.com/espressif/esp-idf/issues/2915#issuecomment-466655373

tripflex, 2019-03-08:
Had to revert all the way down to 2.8.1 just to fix this problem, PLEASE MERGE THE FIX FROM IDF!

KZ-IDEA, 2019-03-08:
Second that.

tripflex, 2019-03-13:
Fixed in 9dba3f497d730230352f79e795f448d64e9eaa24 thank you @rojer

closed 2019-03-13 by tripflex