cesanta/mongoose-os #487: (erro mos build)problemas na contruçao do firmware na plataforma mos.

ALISSOMSJ, 2019-04-03:


ALISSOMSJ, 2019-04-03:
olaa, estou tentando construir o firmware no mos, mas quando envio o comando *mos build* a plataforma envia erro.

$ mos build --platform esp32
Error: /go/src/cesanta.com/mos/build.go:347: code dir ("C:\\Users\\Alisson São João\\Pictures\\app1") should not contain spaces
/go/src/cesanta.com/mos/build_remote.go:38: 
/go/src/cesanta.com/mos/build.go:205: 
/go/src/cesanta.com/mos/build.go:148: 
/go/src/cesanta.com/mos/main.go:176: build failed
exit status 1

Alguém sabe como posso fazer, pois já tenho os drives instalado mas não constrói

DrBomb, 2019-04-03:
The error is clear: `code dir ("C:\Users\Alisson São João\Pictures\app1") should not contain spaces`

You should move your app directory to a place that does not have any spaces.

aleksmarcelo, 2019-04-04:
Alissom vc nao deve usar espaços na definição de  diretórios.