cesanta/mongoose-os #500: mos build esp32

dniadzelka, 2019-05-01:
Hello,

I tried to follow [quickstart](https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md) guide but got errors on **Build app firmware** step

  pc: macOS 10.13.6
  mos version: 2.13.1
  command: **mos build --platform esp32**

`Error: /private/tmp/mos-20190501-29885-7ph270/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/mos/build_remote.go:309: build failed
/private/tmp/mos-20190501-29885-7ph270/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/mos/build.go:208: 
/private/tmp/mos-20190501-29885-7ph270/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/mos/build.go:151: 
/private/tmp/mos-20190501-29885-7ph270/mos-tool-2034cc00c37fee33b300b1c5869b8eeff328fdfa/src/cesanta.com/mos/main.go:178: build failed`

zfeng1, 2019-05-01:
Having the same issue.

- OS: Windows 8.1
- mos version: 2.13.1
- Using mos GUI tool

`App is too big
/mongoose-os/fw/platforms/esp32/Makefile.build:212: recipe for target 'app-bin-check' failed
make: *** [app-bin-check] Error 1
make: *** Waiting for unfinished jobs....
make: Leaving directory '/app'
Error: exit status 2
/go/src/cesanta.com/mos/build_local.go:657: 
/go/src/cesanta.com/mos/build_local.go:644: 
/go/src/cesanta.com/mos/build_local.go:413: 
/go/src/cesanta.com/mos/build.go:208: 
/go/src/cesanta.com/mos/build.go:151: 
/go/src/cesanta.com/mos/main.go:178: build failed
`

Full log attached:
[build.log](https://github.com/cesanta/mongoose-os/files/3135032/build.log)

nliviu, 2019-05-02:
It has been fixed in latest:
```
mos version
The Mongoose OS command line tool
Version: 201905021612
Build ID: 20190502-161444/g6f34b9e-master
Update channel: latest

```

closed 2019-05-02 by nliviu