cesanta/mongoose-os #501: MOS 2.13.0+ BUG - drops sys.mount causing ext_fs to NOT MOUNT!

tripflex, 2019-05-01:
Did clean build on latest and found my ext_fs was not mounting for some reason .. dug into this and found the problem is related to the 2.13.0+ versions of MOS, which for some reason is dropping (or not adding) the `sys.mount` defined in my `mos.yml` file.

Build using 2.12.1 and it works fine ... so anybody using 2.13.0+ and trying to use ext_fs as defined in the docs:
https://mongoose-os.com/docs/mongoose-os/howtos/fs.md

It will NOT work.

`conf0.json` in 2.13.0+:
```json
 "sys": {
  "atca": {
   "i2c_addr": 96,
   "ecdh_slots_mask": 12
  },
  "tz_spec": "EST5EDT4,M3.2.0/02:00:00,M11.1.0/02:00:00",
  "wdt_timeout": 30
 },
```

`conf0.json` in 2.12.1
```json
 "sys": {
  "mount": {
   "path": "/data",
   "dev_type": "esp32part",
   "dev_opts": "{\"label\": \"data\"}",
   "fs_type": "SPIFFS"
  },
  "tz_spec": "EST5EDT4,M3.2.0/02:00:00,M11.1.0/02:00:00",
  "wdt_timeout": 30,
  "atca": {
   "i2c_addr": 96,
   "ecdh_slots_mask": 12
  }
 },
```

tripflex, 2019-05-02:
https://gitter.im/cesanta/mongoose-os?at=5cc847c38446a6023ea0761c

If something is "deprecated" it would be nice if you guys WOULD MENTION THAT IN THE CHANGELOG/RELEASELOG -- more like REMOVED

UGH it's like playing whack-a-mole anytime you guys release updates 😠

nliviu, 2019-06-05:
Has been restored with https://github.com/mongoose-os-libs/fstab/commit/bbcf31da4ed735f7f7ecf1fb82c75eb8dc41a561

closed 2019-06-05 by nliviu