cesanta/mongoose-os #503: There is no connected device on web UI

rid-hrant, 2019-05-12:
Hi. I'm trying to install firmware OS on Esp32 board, but I'm getting error in console, problem occurred on Fedora 29 (Linux kernel 5.0.10-200.fc29.x86_64) and Windows 10. I have already installed drivers for Fedora and Windows.

Mos console log
```
Serial console
20190426-141308/2.13.1-g2034cc0

No serial ports available. Possible reasons:
  - A device is disconnected. Connect and press Ctrl-l.
  - A USB-To-Serial driver is not installed. To install, see instructions.
    When done, restart this tool.
```

Device list
[image]

Driver installation log
```
[hrant@localhost Linux_3.x.x_4.x.x_VCP_Driver_Source]$ make -f Makefile 
make -C /lib/modules/`uname -r`/build M=/home/hrant/Documents/Linux_3.x.x_4.x.x_VCP_Driver_Source modules
make[1]: Entering directory '/usr/src/kernels/5.0.10-200.fc29.x86_64'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/kernels/5.0.10-200.fc29.x86_64'
```

cpq, 2019-05-13:
Do you see a serial device appearing in `/dev` after the driver installation?

Anyhow, this does not sound like a Mongoose OS issue - rather a driver issue.

closed 2019-05-13 by cpq

rid-hrant, 2019-05-13:
> Do you see a serial device appearing in `/dev` after the driver installation?
> 
> Anyhow, this does not sound like a Mongoose OS issue - rather a driver issue.

But I have installed [driver](https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md#3-usb-to-serial-drivers) provided by mongoose. Why it's not working?

cpq, 2019-05-13:
Mongoose OS does not provide any driver.
These links point to the 3rd party places.

rid-hrant, 2019-05-13:
> Mongoose OS does not provide any driver.
> These links point to the 3rd party places.

So what can I do? 😑

cpq, 2019-05-13:
I guess, as usual - search the Internet for solutions.
This project is not the right place though. It assumes that drivers work fine.

rid-hrant, 2019-05-13:
> I guess, as usual - search the Internet for solutions.
> This project is not the right place though. It assumes that drivers work fine.

Oh, You are genius 🙂. Anyway thank you.

nliviu, 2019-05-13:
Are you sure your device has a FTDI USB chip?

rid-hrant, 2019-05-13:
> Are you sure your device has a FTDI USB chip?

@nliviu you are talking about Esp 32 board?

nliviu, 2019-05-13:
Yes.

rid-hrant, 2019-05-13:
> Yes.

> Yes.

I could't find any information about `FTDI`. The board model is [`HiLetgo ESP-WROOM-32 ESP32 ESP-32S`](https://www.amazon.com/gp/product/B0718T232Z/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1).

nliviu, 2019-05-13:
It looks like your device uses a Silabs CP210x USB chip. At least that's what I can see in the picture from the link you provided.
Fedora (5.0.13-300.fc30.x86_64) has all the usual USB/serial drivers:
```
[   69.119648] usbserial: USB Serial support registered for ch341-uart
[   69.120310] ch341 2-2.1:1.0: ch341-uart converter detected
[   69.137471] usb 2-2.1: ch341-uart converter now attached to ttyUSB0
...
[  319.567094] usb 2-2.1: Product: FT232R USB UART
[  319.567095] usb 2-2.1: Manufacturer: FTDI
[  319.626879] usbcore: registered new interface driver ftdi_sio
[  319.627580] usbserial: USB Serial support registered for FTDI USB Serial Device
[  319.628172] ftdi_sio 2-2.1:1.0: FTDI USB Serial Device converter detected
[  319.628309] usb 2-2.1: Detected FT232RL
[  319.638704] usb 2-2.1: FTDI USB Serial Device converter now attached to ttyUSB0
...
[  762.997462] usb 2-2.1: Product: CP2104 USB to UART Bridge Controller
[  762.997487] usb 2-2.1: Manufacturer: Silicon Labs
[  762.997489] usb 2-2.1: SerialNumber: 014C1888
[  763.039487] cp210x 2-2.1:1.0: cp210x converter detected
[  763.056964] usb 2-2.1: cp210x converter now attached to ttyUSB0
```

rid-hrant, 2019-05-13:
@nliviu So I should try to flash via `/dev/ttyUSB0` port?

nliviu, 2019-05-13:
`mos ports` will show you the available devices:
```
 mos ports
/dev/ttyUSB0
```
You might need to add your user to the dialout group to have access to the USB device (google for it).

rid-hrant, 2019-05-13:
> `mos ports` will show you the available devices:
> 
> ```
>  mos ports
> /dev/ttyUSB0
> ```
> 
> You might need to add your user to the dialout group to have access to the USB device as a regular user (google for it).

@nliviu Thank you!