cesanta/mongoose-os #507: LoadStoreError on ESP32 16MB version

T94T, 2019-05-25:
Hi,

I currently got a new ESP32-WROOM-32 with 16MB of flash from Mouser. No matter what firmware (for example the mongoose-os-apps/demo-c) is flashed on the chip I get the error message below. Flashing common esp-idf projects is working fine though.

```
[May 25 11:55:59.384] esp32_wifi.c:442        WiFi AP: SSID EB_81DCC4, channel 6
[May 25 11:56:00.233] I (1773) wifi: Total power save buffer number: 16
[May 25 11:56:00.233] esp32_wifi.c:492        WiFi AP IP: 10.0.0.10/255.255.255.0 gw 10.0.0.10, DHCP range 10.0.0.11 - 10.0.0.111
[May 25 11:56:00.245] esp32_wifi.c:497        WiFi AP: SSID EB_81DCC4, channel 6
[May 25 11:56:00.251] mgos_http_server.c:282  HTTP server started on [80]
[May 25 11:56:00.258] mgos_mdns.c:84          Listening on udp://:5353
[May 25 11:56:00.258] lwip_mdns.c:39          Joining multicast group 224.0.0.251
[May 25 11:56:00.264] mgos_dns_sd.c:426       MDNS initialized, host eduboard.local, ttl 120
[May 25 11:56:00.278]  mgos_rpc_channel_ua:313 0x3ffc9a90 UART0
[May 25 11:56:00.278] main.c:76               MGOS APP Init.
[May 25 11:56:00.284] Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
[May 25 11:56:00.289] Core 0 register dump:
[May 25 11:56:00.289] PC      : 0x4018f163  PS      : 0x00060b30  A0      : 0x800e2c84  A1      : 0x3ffb5170  
[May 25 11:56:00.300] A2      : 0x00000001  A3      : 0x3ffc18c2  A4      : 0x3ffb51a8  A5      : 0x3ffb5190  
[May 25 11:56:00.306] A6      : 0x00000004  A7      : 0x00000000  A8      : 0x3f404c5f  A9      : 0x0000003f  
[May 25 11:56:00.317] A10     : 0x3ffc18b8  A11     : 0x00000034  A12     : 0x00000006  A13     : 0x00000009  
[May 25 11:56:00.322] A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000001f  EXCCAUSE: 0x00000003  
[May 25 11:56:00.328] EXCVADDR: 0x3f404c5f  LBEG    : 0x4018f130  LEND    : 0x4018f143  LCOUNT  : 0x00000000  
[May 25 11:56:00.339] 
[May 25 11:56:00.339] Backtrace: 0x4018f163 0x400e2c81 0x400e6416 0x400e2a7f 0x400e6697 0x40083b0c
[May 25 11:56:00.345] 
[May 25 11:56:00.345] --- BEGIN CORE DUMP ---
```

henrikengelbrink, 2020-03-09:
I have the same issue now. @T94T did you found the solution for this problem ?

cpq, 2020-03-09:
Build your app from a command line, then `mos flash && mos console`. The `mos console` command starts a core backtrace on crash - the only thing it needs is an `.elf` file. The backtrace should give a clue what is going on.

henrikengelbrink, 2020-03-09:
Thank you for your answer. I've done this and I'm getting these log files, but I'm new to mongoose os and I'm totally lost finding any helpful information in these cryptographic logs.

DrBomb, 2020-03-09:
@henrikengelbrink 

I've been using ESP32 with mongoose-os with 16M flash sizes without any issue.

Please do as @cpq outlined:

1. Make sure you're running the latest version of the mos tool
2. Make a clean local build.
3. Run `mos flash` and `mos console` to flash and start the serial monitor.
4. Watch your app closely. In the case of a core dump, the serial monitor will detect the crash and capture the output of the core dump.
5. After capturing the core dump, the debugger will be invoked automatically. You can use the `bt` command on gdb to get the stack trace
6. Please post the trace here.

Feel free to ask more questions on the project's gitter channel https://gitter.im/cesanta/mongoose-os

henrikengelbrink, 2020-03-09:
@DrBomb Thank you for your detailed information.

I already found the problem. The `.protocol` parameter in the ap config struct was missing. This seems to be a new thing and it was not mentioned in the docs. Would be helpful if you can update the docs that nobody will have the same issue in the future.

```
struct mgos_config_wifi_ap ap_config = {
    .enable = true,
    .ssid = "esp32eh",
    .pass = "wilano123",
    .ip = "192.168.99.1",
    .netmask = "255.255.255.0",
    .dhcp_start = "192.168.99.20",
    .dhcp_end = "192.168.99.100",
    .channel = 5,
    .max_connections = 3,
    .protocol = "BGN" // <- was missing
  };
  bool wifiSuccess = mgos_wifi_setup_ap(&ap_config);
  LOG(LL_INFO, ("Wifi Success: %d", wifiSuccess));
```

This code is working now and the AP is running and it's possible to connect to it.

DrBomb, 2020-03-09:
Ok, so your problem was not because your module was 16Mb but because your application was crashing due to application code.

I would suggest just using the config parameters for AP config instead of initializing the AP mode yourself. But if it works that's good.

I'll be closing this issue because it is 1 year old already.

closed 2020-03-09 by DrBomb