cesanta/mongoose-os #511: Option to use REF_TICK for uart clock source on ESP32 (corrupted baud rate with DFS)

pwiner99, 2019-08-09:
When dynamic frequency scaling (DFS) is enabled on ESP32 the APB clock frequency feeding the uart can change resulting in incorrect baud rate.  This problem can be seen by setting CONFIG_PM_ENABLE=y CONFIG_PM_DFS_INIT_AUTO=y in the build vars for ESP_IDF_SDKCONFIG_OPTS.

Suggest this line https://github.com/cesanta/mongoose-os/blob/787ac38aae90231b9db9c71cf19dfc617ca58166/platforms/esp32/src/esp32_uart.c#L407

be modified to something like:

uint32_t conf0 = cfg->dev.use_ref_tick ? 0 : UART_TICK_REF_ALWAYS_ON;

where use_ref_tick is added to the esp hal and exposed with a config-schema.

rojer, 2020-04-12:
REF_TICK is only 1 MHz though, and UART required at least 1x divider, so it's only suitable for low baud rates

Harvie, 2020-10-27:
in `deps/modules/mongoose-os/platforms/esp32/src/esp32_uart.c` i changed on line 429:

`uint32_t conf0 = UART_TICK_REF_ALWAYS_ON;` to `uint32_t conf0 = 1;`

Now the baudrate is incorrect, so `mos console` shows only whitespaces.

rojer, 2020-10-27:
fixed in https://github.com/cesanta/mongoose-os/commit/5b937f05a949c6227b6a17267533f490f4b8c661. baud rates up to 460800 verified to be working with ref_tick, for higher rates we fall back to APB.

closed 2020-10-27 by rojer

Harvie, 2020-10-27:
Perfect! Thank you :-)

Harvie, 2020-10-27:
I've just tested and it fixed the problem. Here is easy way to test it in any mongoose project:

```bash
curl 'https://raw.githubusercontent.com/cesanta/mongoose-os/master/platforms/esp32/src/esp32_uart.c' > ./deps/modules/mongoose-os/platforms/esp32/src/esp32_uart.c

#revert back to 2.18
curl 'https://raw.githubusercontent.com/cesanta/mongoose-os/2.18.0/platforms/esp32/src/esp32_uart.c' > ./deps/modules/mongoose-os/platforms/esp32/src/esp32_uart.c
```