cesanta/mongoose-os #514: Got warning "Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it."

dimitrov-adrian, 2019-09-21:
With latest builds I got next warning, any clue what it meaning and how to solve?

```
[Sep 21 11:51:21.853] I (4) cpu_start: Chip Revision: 1
[Sep 21 11:51:21.857] W (4) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.
```

rojer, 2019-09-21:
we cannot make it the default because we need this to work with older chip revisions.
but you can override sdkconfig options in mos.yml file:

```
build_vars:
  ESP_IDF_SDKCONFIG_OPTS: >
    ${build_vars.ESP_IDF_SDKCONFIG_OPTS}
    CONFIG_ESP32_REV_MIN=1
```

closed 2019-09-21 by rojer

dimitrov-adrian, 2020-01-22:
Just a note that @rojer seems to doesn't works, but found next as working

```
build_vars:
  ESP_IDF_SDKCONFIG_OPTS: >
    ${build_vars.ESP_IDF_SDKCONFIG_OPTS}
    CONFIG_ESP32_REV_MIN=1
    CONFIG_ESP32_REV_MIN_0=n
    CONFIG_ESP32_REV_MIN_1=y
```