cesanta/mongoose-os #515: Quickstart SIGSEGV error on 'mos build' for ESP8266 on Arch

lhanson, 2019-10-12:
I'd love to start using Mongoose OS instead of the Arduino IDE, but I'm blocked at building a sample project. Steps to reproduce, following the [quickstart](https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md):

- Install mos tool for Arch Linux.
- Plug in my new [ESP8266](https://www.amazon.com/gp/product/B07V9DWV6T) board
- Launch `mos` tool
- Create new app in web UI `mos clone https://github.com/mongoose-os-apps/demo-js app1`
- The port is correctly identified as `/dev/ttyUSB0`, and I select `ESP8266` in the board dropdown.
[image]

- Type `mos build` into the command input and get the following error:

```$ mos build --platform esp8266
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x47 pc=0xf764e746]

runtime stack:
runtime.throw(0x8ad7a34, 0x2a)
	/usr/lib/go-1.10/src/runtime/panic.go:616 +0x71
runtime.sigpanic()
	/usr/lib/go-1.10/src/runtime/signal_unix.go:372 +0x239

goroutine 37 [syscall]:
runtime.cgocall(0x87f8e30, 0x1af36ee8, 0x8ad6357)
	/usr/lib/go-1.10/src/runtime/cgocall.go:128 +0x75 fp=0x1af36ecc sp=0x1af36eb4 pc=0x8049f35
net._C2func_getaddrinfo(0x1af0a1c0, 0x0, 0x1af08520, 0x1af0c0c0, 0x0, 0x0, 0x0)
	_cgo_gotypes.go:85 +0x4f fp=0x1af36ee8 sp=0x1af36ecc pc=0x81a153f
net.cgoLookupIPCNAME.func1(0x1af0a1c0, 0x0, 0x1af08520, 0x1af0c0c0, 0x10, 0x10, 0x0)
	/usr/lib/go-1.10/src/net/cgo_unix.go:149 +0x107 fp=0x1af36f10 sp=0x1af36ee8 pc=0x81a6e97
net.cgoLookupIPCNAME(0x1af10040, 0xf, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/lib/go-1.10/src/net/cgo_unix.go:149 +0x155 fp=0x1af36f90 sp=0x1af36f10 pc=0x81a27e5
net.cgoIPLookup(0x1af16380, 0x1af10040, 0xf)
	/usr/lib/go-1.10/src/net/cgo_unix.go:201 +0x2d fp=0x1af36fe0 sp=0x1af36f90 pc=0x81a2d4d
runtime.goexit()
	/usr/lib/go-1.10/src/runtime/asm_386.s:1665 +0x1 fp=0x1af36fe4 sp=0x1af36fe0 pc=0x8098551
created by net.cgoLookupIP
	/usr/lib/go-1.10/src/net/cgo_unix.go:211 +0x8c

goroutine 1 [runnable]:
syscall.Syscall(0xdc, 0x6, 0x1af86000, 0x1000, 0x0, 0x1000, 0x0)
	/usr/lib/go-1.10/src/syscall/asm_linux_386.s:20 +0x5
syscall.Getdents(0x6, 0x1af86000, 0x1000, 0x1000, 0xc, 0x1add76b8, 0x1)
	/usr/lib/go-1.10/src/syscall/zsyscall_linux_386.go:482 +0x4d
syscall.ReadDirent(0x6, 0x1af86000, 0x1000, 0x1000, 0x50, 0x1af86050, 0x1af88000)
	/usr/lib/go-1.10/src/syscall/syscall_linux.go:772 +0x39
internal/poll.(*FD).ReadDirent(0x1af80000, 0x1af86000, 0x1000, 0x1000, 0x0, 0x0, 0x0)
	/usr/lib/go-1.10/src/internal/poll/fd_unix.go:404 +0x9a
os.(*File).readdirnames(0x1af7e000, 0xfffffffe, 0x1add7748, 0x8, 0x89975c0, 0x89975c0, 0x80d487a)
	/usr/lib/go-1.10/src/os/dir_unix.go:67 +0x159
os.(*File).Readdirnames(0x1af7e000, 0xffffffff, 0x8af214c, 0x8b9a1d0, 0x1af82000, 0x1000006, 0x1af7e000)
	/usr/lib/go-1.10/src/os/dir.go:45 +0x2d
os.(*File).readdir(0x1af7e000, 0xffffffff, 0x0, 0x0, 0x1af7e000, 0x0, 0x0)
	/usr/lib/go-1.10/src/os/dir_unix.go:24 +0x4c
os.(*File).Readdir(0x1af7e000, 0xffffffff, 0x1af7e000, 0x0, 0x0, 0x0, 0x0)
	/usr/lib/go-1.10/src/os/dir.go:26 +0x2d
io/ioutil.ReadDir(0x1af300a0, 0x1d, 0x800001ed, 0x0, 0x0, 0x1af72000, 0x2)
	/usr/lib/go-1.10/src/io/ioutil/ioutil.go:101 +0x55
github.com/mongoose-os/mos/common/ourio.CopyDir(0x1af300a0, 0x1d, 0x1ac9d080, 0x2b, 0x1ae74560, 0x2, 0x2, 0x29, 0x1ac000e0)
	/go/src/github.com/mongoose-os/mos/common/ourio/copy.go:165 +0x291
github.com/mongoose-os/mos/common/ourio.CopyDir(0x1ae35f60, 0x1a, 0x1ac9cc00, 0x28, 0x1ae74560, 0x2, 0x2, 0x0, 0x8968460)
	/go/src/github.com/mongoose-os/mos/common/ourio/copy.go:184 +0x51a
main.buildRemote(0x1ae9bb80, 0x0, 0x0)
	/go/src/github.com/mongoose-os/mos/mos/build_remote.go:79 +0x193
main.doBuild(0x8b9f5c0, 0x1ac9e01c, 0x1ae9bb80, 0x0, 0x0)
	/go/src/github.com/mongoose-os/mos/mos/build.go:220 +0xc5b
main.buildHandler(0x8b9f5c0, 0x1ac9e01c, 0x0, 0x0, 0x0, 0x0)
	/go/src/github.com/mongoose-os/mos/mos/build.go:166 +0xc1
main.run(0x1ae23b80, 0x8b9f5c0, 0x1ac9e01c, 0x0, 0x0, 0x0, 0x90afa93)
	/go/src/github.com/mongoose-os/mos/mos/main.go:195 +0x78
main.main()
	/go/src/github.com/mongoose-os/mos/mos/main.go:297 +0x506

goroutine 19 [chan receive]:
github.com/mongoose-os/mos/vendor/github.com/golang/glog.(*loggingT).flushDaemon(0x91497a0)
	/go/src/github.com/mongoose-os/mos/vendor/github.com/golang/glog/glog.go:882 +0x6b
created by github.com/mongoose-os/mos/vendor/github.com/golang/glog.init.0
	/go/src/github.com/mongoose-os/mos/vendor/github.com/golang/glog/glog.go:410 +0x1d4

goroutine 21 [select]:
github.com/mongoose-os/mos/vendor/go.opencensus.io/stats/view.(*worker).start(0x1ac99b20)
	/go/src/github.com/mongoose-os/mos/vendor/go.opencensus.io/stats/view/worker.go:147 +0xfd
created by github.com/mongoose-os/mos/vendor/go.opencensus.io/stats/view.init.0
	/go/src/github.com/mongoose-os/mos/vendor/go.opencensus.io/stats/view/worker.go:29 +0x4b

goroutine 22 [syscall]:
os/signal.signal_recv(0x0)
	/usr/lib/go-1.10/src/runtime/sigqueue.go:139 +0x12f
os/signal.loop()
	/usr/lib/go-1.10/src/os/signal/signal_unix.go:22 +0x1a
created by os/signal.init.0
	/usr/lib/go-1.10/src/os/signal/signal_unix.go:28 +0x37

goroutine 23 [sleep]:
time.Sleep(0x5f5e100, 0x0)
	/usr/lib/go-1.10/src/runtime/time.go:102 +0x15b
main.main.func1()
	/go/src/github.com/mongoose-os/mos/mos/main.go:233 +0x29
created by main.main
	/go/src/github.com/mongoose-os/mos/mos/main.go:232 +0x163

goroutine 24 [select]:
net/http.(*Transport).getConn(0x90aa820, 0x1af08160, 0x0, 0x1af12080, 0x5, 0x1af10040, 0x13, 0x0, 0x0, 0x0)
	/usr/lib/go-1.10/src/net/http/transport.go:962 +0x469
net/http.(*Transport).RoundTrip(0x90aa820, 0x1af0e080, 0x90aa820, 0x0, 0x0)
	/usr/lib/go-1.10/src/net/http/transport.go:409 +0x4bd
net/http.send(0x1af0e080, 0x8b98930, 0x90aa820, 0x0, 0x0, 0x0, 0x0, 0x0, 0x9, 0x2, ...)
	/usr/lib/go-1.10/src/net/http/client.go:252 +0x151
net/http.(*Client).send(0x1af08000, 0x1af0e080, 0x0, 0x0, 0x0, 0x0, 0x0, 0x2, 0x80571f9, 0x1, ...)
	/usr/lib/go-1.10/src/net/http/client.go:176 +0xaa
net/http.(*Client).Do(0x1af08000, 0x1af0e080, 0xa, 0x1af1a000, 0x42)
	/usr/lib/go-1.10/src/net/http/client.go:615 +0x23c
github.com/mongoose-os/mos/mos/update.GetServerMosVersion(0x8aaccc9, 0x7, 0x1ac437d4, 0x2, 0x2, 0x0, 0x0, 0x0)
	/go/src/github.com/mongoose-os/mos/mos/update/update.go:88 +0x20e
main.doBuild.func1(0x1ae9bb80, 0x1ae23bc0)
	/go/src/github.com/mongoose-os/mos/mos/build.go:183 +0xc2
created by main.doBuild
	/go/src/github.com/mongoose-os/mos/mos/build.go:182 +0xd2

goroutine 34 [select]:
net.(*Resolver).LookupIPAddr(0x9148160, 0x8b9f5e0, 0x1af16200, 0x1af10040, 0xf, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/lib/go-1.10/src/net/lookup.go:212 +0x459
net.(*Resolver).internetAddrList(0x9148160, 0x8b9f5e0, 0x1af16200, 0x8aa8fbc, 0x3, 0x1af10040, 0x13, 0x0, 0x0, 0x0, ...)
	/usr/lib/go-1.10/src/net/ipsock.go:293 +0x460
net.(*Resolver).resolveAddrList(0x9148160, 0x8b9f5e0, 0x1af16200, 0x8aa9632, 0x4, 0x8aa8fbc, 0x3, 0x1af10040, 0x13, 0x0, ...)
	/usr/lib/go-1.10/src/net/dial.go:193 +0x4ae
net.(*Dialer).DialContext(0x1aca0180, 0x8b9f5c0, 0x1ac9e01c, 0x8aa8fbc, 0x3, 0x1af10040, 0x13, 0x0, 0x0, 0x0, ...)
	/usr/lib/go-1.10/src/net/dial.go:375 +0x20d
net.(*Dialer).DialContext-fm(0x8b9f5c0, 0x1ac9e01c, 0x8aa8fbc, 0x3, 0x1af10040, 0x13, 0x804b744, 0x40, 0x823bb79, 0x1ac9e01c)
	/usr/lib/go-1.10/src/net/http/transport.go:46 +0x50
net/http.(*Transport).dial(0x90aa820, 0x8b9f5c0, 0x1ac9e01c, 0x8aa8fbc, 0x3, 0x1af10040, 0x13, 0x8af20ac, 0x1f, 0xffffffe1, ...)
	/usr/lib/go-1.10/src/net/http/transport.go:898 +0x178
net/http.(*Transport).dialConn(0x90aa820, 0x8b9f5c0, 0x1ac9e01c, 0x0, 0x1af12080, 0x5, 0x1af10040, 0x13, 0x0, 0x0, ...)
	/usr/lib/go-1.10/src/net/http/transport.go:1143 +0x281
net/http.(*Transport).getConn.func4(0x90aa820, 0x8b9f5c0, 0x1ac9e01c, 0x1af08180, 0x1af12100)
	/usr/lib/go-1.10/src/net/http/transport.go:957 +0x3e
created by net/http.(*Transport).getConn
	/usr/lib/go-1.10/src/net/http/transport.go:956 +0x2be

goroutine 36 [select]:
net.cgoLookupIP(0x8b9f5a0, 0x1af08220, 0x1af10040, 0xf, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/lib/go-1.10/src/net/cgo_unix.go:212 +0x138
net.(*Resolver).lookupIP(0x9148160, 0x8b9f5a0, 0x1af08220, 0x1af10040, 0xf, 0x1af080c0, 0x1ac3667c, 0x1ac36600, 0x0, 0x0)
	/usr/lib/go-1.10/src/net/lookup_unix.go:95 +0xca
net.(*Resolver).(net.lookupIP)-fm(0x8b9f5a0, 0x1af08220, 0x1af10040, 0xf, 0x4, 0x1af080c0, 0x0, 0x1af36748, 0x0)
	/usr/lib/go-1.10/src/net/lookup.go:192 +0x40
net.glob..func10(0x8b9f5a0, 0x1af08220, 0x1af0c088, 0x1af10040, 0xf, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/lib/go-1.10/src/net/hook.go:19 +0x3c
net.(*Resolver).LookupIPAddr.func1(0x0, 0x0, 0x0, 0x0)
	/usr/lib/go-1.10/src/net/lookup.go:206 +0xb5
internal/singleflight.(*Group).doCall(0x9148680, 0x1af20090, 0x1af10040, 0xf, 0x1af08240)
	/usr/lib/go-1.10/src/internal/singleflight/singleflight.go:95 +0x21
created by internal/singleflight.(*Group).DoChan
	/usr/lib/go-1.10/src/internal/singleflight/singleflight.go:88 +0x266
exit status 2
Command completed.```

rojer, 2019-10-12:
yeah, "generic" linux binaries are busted because of library incompatibilities.
you'll need to build yourself a package, see here - https://github.com/mongoose-os/mos#installing-on-arch-linux

lhanson, 2019-10-12:
I did build the package myself following those instructions. I got several errors for missing dependencies, which I installed via `pacman` manually. IIRC those included `jshon` and `libftdi-compat`. Are there specific versions of those libraries I need to pin the build to?

rojer, 2019-10-12:
not really. if you can modify the package definition to make it so they are fetched automatically, please send a PR. as you can probably tell, we're not arch users here :)

rojer, 2019-10-12:
i'm going to close this issue, as it's a known problem already.

closed 2019-10-12 by rojer

lhanson, 2019-10-13:
> if you can modify the package definition to make it so they are fetched automatically, please send a PR

Sure... since this is apparently a known problem, is there another open issue I can reference? I'm not quite sure what to make of the fact that he build need to "fetch the dependencies automatically". Are the dependencies in the package too old? Wrong versions? It sounds like there is already some knowledge here that I could leverage rather than start from scratch.