cesanta/mongoose-os #521: mos wifi fails - Error: write /dev/cu.SLAB_USBtoUART: file already closed on MacOS

perbu, 2019-12-12:
Hi.

Host OS: MacOS Catalina 10.15.2
MOS: 2.16.0 (2.16.0~brew)
Device: ESP32Dev (TTGO wrover32 with a CP2104)

Start out with a blank directory. Copy a minimal mos.yml:
```
author: mongoose-os
# It is here mainly for use in tests to make sure this is possible.
description: Most minimal app - no libs, no networking, no RPC, nothing
version: 1.0
manifest_version: 2017-05-18

libs_version: ${mos_version}
modules_version: ${mos_version}
mongoose_os_version: ${mos_version}

sources:
  - src

tags:
  - c

libs:
  - origin: https://github.com/mongoose-os-libs/boards
  - origin: https://github.com/mongoose-os-libs/core
```

Build an image:
```
➜  blank mos --platform esp32 build
Connecting to https://mongoose.cloud, user test
Uploading sources (1065 bytes)
Firmware saved to /Users/perbu/esp32/mongoose/blank/build/fw.zip
```
Flash the image:
```
mos --port /dev/cu.SLAB_USBtoUART flash
Loaded blank/esp32 version 1.0 (20191212-125043)
Opening /dev/cu.SLAB_USBtoUART @ 115200...
Connecting to ESP32 ROM, attempt 1 of 10...
  Connected, chip: ESP32D0WDQ5 R1
Running flasher @ 921600...
  Flasher is running
Flash size: 4194304, params: 0x022f (dio,32m,80m)
Flash encryption: disabled, scheme: None
Secure boot: disabled
Deduping...
    23152 @ 0x1000 -> 6768
     3072 @ 0x8000 -> 0
    16384 @ 0x9000 -> 0
     8192 @ 0xd000 -> 0
   262144 @ 0x190000 -> 0
Writing...
     4096 @ 0x1000
     4096 @ 0x6000
   897024 @ 0x10000
Wrote 903440 bytes in 10.10 seconds (699.14 KBit/sec)
Verifying...
    23152 @ 0x1000
     3072 @ 0x8000
    16384 @ 0x9000
     8192 @ 0xd000
   896672 @ 0x10000
   262144 @ 0x190000
Booting firmware...
All done!
```
Then I try to set the wifi credentials:
```
➜  blank mos --port /dev/cu.SLAB_USBtoUART wifi $SSID $PASSWORD
Getting configuration...
Error: write /dev/cu.SLAB_USBtoUART: file already closed
/private/tmp/mos-20191107-31823-1o4s33s/mos-adcad3f2ca90789fc5426dc8c6302383effaa415/src/cesanta.com/go/src/github.com/mongoose-os/mos/common/mgrpc/codec/serial.go:203: 
/private/tmp/mos-20191107-31823-1o4s33s/mos-adcad3f2ca90789fc5426dc8c6302383effaa415/src/cesanta.com/go/src/github.com/mongoose-os/mos/common/mgrpc/codec/serial.go:262: 
/private/tmp/mos-20191107-31823-1o4s33s/mos-adcad3f2ca90789fc5426dc8c6302383effaa415/src/cesanta.com/go/src/github.com/mongoose-os/mos/common/mgrpc/codec/stream.go:331: 
/private/tmp/mos-20191107-31823-1o4s33s/mos-adcad3f2ca90789fc5426dc8c6302383effaa415/src/cesanta.com/go/src/github.com/mongoose-os/mos/common/mgrpc/mgrpc.go:401: 
/private/tmp/mos-20191107-31823-1o4s33s/mos-adcad3f2ca90789fc5426dc8c6302383effaa415/src/cesanta.com/go/src/github.com/mongoose-os/mos/mos/dev/dev_conn_impl.go:167: 
/private/tmp/mos-20191107-31823-1o4s33s/mos-adcad3f2ca90789fc5426dc8c6302383effaa415/src/cesanta.com/go/src/github.com/mongoose-os/mos/mos/dev/dev_conn_impl.go:190: 
/private/tmp/mos-20191107-31823-1o4s33s/mos-adcad3f2ca90789fc5426dc8c6302383effaa415/src/cesanta.com/go/src/github.com/mongoose-os/mos/mos/dev/dev_conn.go:117: 
/private/tmp/mos-20191107-31823-1o4s33s/mos-adcad3f2ca90789fc5426dc8c6302383effaa415/src/cesanta.com/go/src/github.com/mongoose-os/mos/mos/config/config.go:86: 
/private/tmp/mos-20191107-31823-1o4s33s/mos-adcad3f2ca90789fc5426dc8c6302383effaa415/src/cesanta.com/go/src/github.com/mongoose-os/mos/mos/main.go:196: wifi failed
```

I've tried three different cabels, tried rebooting, two different USB-C bridges.

The console works with screen: 
- screen /dev/cu.SLAB_USBtoUART 115200
- mos --port /dev/cu.SLAB_USBtoUART console

Verbose logs:
```
mos --v 4 --logtostderr --platform esp32 --port /dev/cu.SLAB_USBtoUART wifi $SSID $PASSWORD
I1212 14:04:34.797356    1091 main.go:253] Version: 2.16.0
I1212 14:04:34.797469    1091 main.go:254] Build ID: 2.16.0~brew
I1212 14:04:34.797514    1091 main.go:255] Update channel: release
I1212 14:04:34.797818    1091 mgrpc.go:237] Connecting to /dev/cu.SLAB_USBtoUART over tSerial
I1212 14:04:34.797850    1091 serial.go:83] Opening /dev/cu.SLAB_USBtoUART...
I1212 14:04:34.813714    1091 serial.go:98] /dev/cu.SLAB_USBtoUART opened: &{0xc00032e0c0 115200}, err: <nil>
Getting configuration...
I1212 14:04:34.833725    1091 ourutil.go:32] Getting configuration...
I1212 14:04:34.833733    1091 mgrpc.go:320] Started recv loop, codec: [streamConnectionCodec to /dev/cu.SLAB_USBtoUART]
I1212 14:04:34.833835    1091 mgrpc.go:322] recv ...
I1212 14:04:34.835835    1091 mgrpc.go:397] created a request with id 1640665524183
I1212 14:04:34.836228    1091 serial.go:260] sending handshake...
I1212 14:04:34.836267    1091 serial.go:186] sent 9 "\"\"\"\x04\"\"\"\n\n"
I1212 14:04:35.034252    1091 stream.go:292] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I1212 14:04:35.046064    1091 serial.go:260] sending handshake...
I1212 14:04:35.046098    1091 serial.go:186] sent 9 "\"\"\"\x04\"\"\"\n\n"
I1212 14:04:35.235399    1091 stream.go:292] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I1212 14:04:35.254362    1091 serial.go:260] sending handshake...
I1212 14:04:35.254395    1091 serial.go:186] sent 9 "\"\"\"\x04\"\"\"\n\n"
(..)
```

I've tried googling and all I can think of.

DrBomb, 2019-12-12:
`file already closed` seems like a port-already-in-use error. But If you say the console works it does not make much sense. I'm not familiar with the macOS version either.

I would suggest first updating the mos binary to latest or rolling back to release. Use `mos update latest` and `mos update release` to do that. To see if both fail the same.

You can also make the wifi changes via `mos config-set`. This would be a way to set it up: `mos config-set wifi.sta.enable=true wifi.sta.ssid=YOUR_SSID wifi.sta.pass=YOURWIFIPASS` That should work.

perbu, 2019-12-13:
@DrBomb I've tried both the release mos as well as mos-latest. Same error. I don't have my device with me atm so I can't test config-set, but I've gotten similar errors when I do "mos ls". I'll do some more testing this weekend, perhaps with a different device. I might find some wemos d1 in a drawer somewhere and it'll be an interesting data point.

I agree that this smell like some sort of locking error. The weird thing is that "mos flash" and "mos console!" works every time, so I'm guessing there is some funkyness in the way the serial communication is initialized in parts of the code. Perhaps tied to some RPC mechanism.

jakimcho, 2019-12-16:
Hello
I'm experiencing the same problem. I'm able to flash, but configuring the wifi fails almost every time.
I found some threads yesterday that the WiFi feature needs greater current to be configured. There are some problematic USB cables that cannot supply that current and the operation fails.
I have also one NodeMCU module that worked like a charm but it is 8266 ESP.

My suggestion is to try with different USB cables. Also since you are using a MAC book, make sure it is plugged to the main power supply.

perbu, 2019-12-18:
I think the power envelope idea is moot. Flashing works and mos console works - even though the little thing is running an accesspoint. Also the following commands also fail:

mos ls - fails
mos get - fails
mos rm - fails
mos config-get - fails

So basically "mos console and mos flash" work fine but things that I would speculate are using RPC are failing.

nliviu, 2019-12-18:
You need
```
  - origin: https://github.com/mongoose-os-libs/rpc-service-config
  - origin: https://github.com/mongoose-os-libs/rpc-service-fs
  - origin: https://github.com/mongoose-os-libs/rpc-uart
  - origin: https://github.com/mongoose-os-libs/wifi
```
for `mos config-set`, `mos ls`, `mos wifi` over serial connection.

perbu, 2019-12-22:
@nliviu Of course. That makes perfect sense.  Thanks!

I think it might be a good idea to try to detect this and issue a warning when the user tries to do something that requires an RPC endpoint to be present on the mcu and we get these short reads.

Perhaps I can come up with a suggested patch if  I can find my way around the codebase.

Cheers.

closed 2019-12-22 by nliviu

rojer, 2019-12-22:
@perbu, added a warning - https://github.com/mongoose-os/mos/commit/5c1464cb0558ab4a8cb74b7cfa9d849c338fc281

perbu, 2019-12-23:
@rojer That is great. 

I know I'm now the only one who has struggled with this. I've found 10-20 threads in various forums which have all been without a concrete answer so this will surely be appreciated.

Merry Christmas.

bigoper, 2020-05-12:
@perbu @rojer 
Hey Guys,
I hope I could use your help with this issue.

### ISSUE
I'm following your notes and I'm unable to reproduce (solve the problem)

### ERROR
```
$ mos wifi "op6" "Bigoper@0!9"
Getting configuration...
E0511 20:05:45.990884   68498 serial.go:273] No response to handshake. Is /dev/cu.wchusbserial14240 the right port? Is rpc-uart enabled?
E0511 20:05:51.197750   68498 serial.go:273] No response to handshake. Is /dev/cu.wchusbserial14240 the right port? Is rpc-uart enabled?
E0511 20:05:56.405037   68498 serial.go:273] No response to handshake. Is /dev/cu.wchusbserial14240 the right port? Is rpc-uart enabled?
Error: write /dev/cu.wchusbserial14240: file already closed
/private/tmp/mos-20200511-56362-y6vyca/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/common/mgrpc/codec/serial.go:206: 
/private/tmp/mos-20200511-56362-y6vyca/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/common/mgrpc/codec/serial.go:265: 
/private/tmp/mos-20200511-56362-y6vyca/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/common/mgrpc/codec/stream.go:331: 
/private/tmp/mos-20200511-56362-y6vyca/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/common/mgrpc/mgrpc.go:401: 
/private/tmp/mos-20200511-56362-y6vyca/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/cli/dev/dev_conn_impl.go:167: 
/private/tmp/mos-20200511-56362-y6vyca/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/cli/dev/dev_conn_impl.go:190: 
/private/tmp/mos-20200511-56362-y6vyca/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/cli/dev/dev_conn.go:117: 
/private/tmp/mos-20200511-56362-y6vyca/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/cli/config/config.go:86: 
/private/tmp/mos-20200511-56362-y6vyca/mos-18c19395db3292985b1d067c4188efd91eea62d3/go/src/github.com/mongoose-os/mos/cli/main.go:197: wifi failed
exit status 1
```

> mos.yml  
```
author: mongoose-os
description: A JS-enabled demo Mongoose OS firmware
# arch: PLATFORM
version: 1.0
manifest_version: 2017-05-18
libs_version: ${mos.version}
modules_version: ${mos.version}
mongoose_os_version: ${mos.version}

config_schema:
  - ["mqtt.server", "iot.eclipse.org:1883"]
  - ["i2c.enable", true]
  - ["blinker.pin", "i", 5, {title: "GPIO pin an LED blinker is attached to"}]

tags:
  - js

filesystem:
  - fs

libs:
  - origin: https://github.com/mongoose-os-libs/boards
  - origin: https://github.com/mongoose-os-libs/js-demo-bundle
  # https://github.com/cesanta/mongoose-os/issues/521#issuecomment-567168538
  - origin: https://github.com/mongoose-os-libs/rpc-service-config
  - origin: https://github.com/mongoose-os-libs/rpc-service-fs
  - origin: https://github.com/mongoose-os-libs/rpc-uart
  - origin: https://github.com/mongoose-os-libs/wifi
  # - origin: https://github.com/mongoose-os-libs/mqtt
  # no need for ^ if 'js-demo-bundle' is added
```

> init.js
```
load('api_gpio.js'); // load is a special function in Mongoose OS
load('api_timer.js');

load('api_mqtt.js');
load('api_gpio.js');

let pin = 0, topic = 'my/topic';

GPIO.set_button_handler(pin, GPIO.PULL_UP, GPIO.INT_EDGE_NEG, 200, function() {
  MQTT.pub('my/topic', JSON.stringify({ a: 1, b: 2 }));
}, null);

MQTT.sub('my/topic/#', function(conn, topic, msg) {
  print('Topic:', topic, 'message:', msg);
}, null);

/**
 * [led description]
 * Documentation
 * - https://github.com/playelek/pinout-doit-32devkitv1/blob/master/pinoutDOIT32devkitv1.png
 * - https://www.espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf
 * - GPIO Pin Definition - page 14
 * @var {[type]}
 */
let led1 = 1; //GPIO1 - PIN TXD0
let led4 = 4; //GPIO4 - PIN D2

// Blink built-in LED every second
GPIO.set_mode(led1, GPIO.MODE_OUTPUT);
GPIO.set_mode(led4, GPIO.MODE_OUTPUT);
Timer.set(100 /* 1 sec */ , true /* repeat */ , function() {
 print("Toggling LED");
 let value4 = GPIO.toggle(led4);
}, null);
Timer.set(500 /* 1 sec */ , true /* repeat */ , function() {
  print("Toggling LED");
 let value1 = GPIO.toggle(led1);
}, null);
```

Thanks in advance,
Avi.

bigoper, 2020-05-12:
### UPDATE (SOLVED)
Setting GPIO1 > PIN TXD0 as OUTPUT was the root cause.
> using a diff PIN resolved that.