cesanta/mongoose-os #522: WPA3

genikineg, 2019-12-16:
Will the support WPA3 ?

rojer, 2019-12-16:
if/when vendors add support to their SDKs.

somerando905, 2021-05-19:
Bumpity bump?

Espressif recently added WPA3 support in ESP8266-RTOS-SDK. I believe ESP-IDF has had WPA3 support since about a year ago.

Perhaps more crucially, both SDKs now support Protected Management Frames which help prevent deauthentication attacks and can also be used with WPA2.

psiberfunk, 2022-08-12:
Do you still use the non-OS SDK , even for running on ESP32 chips ?  Are there plans to convert over to the RTOS version?

rojer, 2022-08-12:
there is no such thing as non-os sdk for ESP32. we use ESP-IDF, which is based on RTOS.
for ESP8266 we are still using non-os SDK and there are no immediate plans to make the switch to RTOS sdk.

psiberfunk, 2022-08-12:
So does the ESP32 then support WPA3 on mongoose-os ?