cesanta/mongoose-os #525: How to store uint32_t in mos.yml config_schema

Harvie, 2020-01-10:
Hello,
is there list of config_schema data types? If i use it to store uint32_t number like this, it gets truncated to half:

`  - ["number", "i", 4294967295, {title: "big number"}]`

How can i save such number?

rojer, 2020-01-10:
the types are i for int, b for bool (but int type is used to represent, with values of 0 and 1), s for string and d for double.

ti represent 4294967295 is 0xffffffff and is -1 as a signed int.
so use -1 in the config and then cast to uint32_t.

closed 2020-01-10 by rojer

Harvie, 2020-01-10:
But i don't need only 4294967295 which translates to -1. I need to store all numbers from 0 to 4294967295...

rojer, 2020-01-10:
the  for the moment you will have to go through the trouble of using the negative representation.
4294967294 is -2 and so on.

or you can use double for your config value type.

Harvie, 2020-01-10:
> or you can use double for your config value type.

Yes. Perfect, this works. Sorry i didn't noticed the first time you mentioned the "d" type...
Still would be nice to have unsigned types in case i wanted to store higher numbers :-)

Anyway, this did the trick. Thank you very much! :-)

I still have one minor issue. I have my variable in mos.yml of my module and i update it like this: `mgos_sys_config_set_my_variable(123);`

But it does not survive the reboot of the device. I want to make it persistent. How do i save it to flash?
I saw the docs here: https://github.com/cesanta/mongoose-os-docs/blob/master/mongoose-os/api/core/mgos_sys_config.h.md

And figured i should do something like this to commit the changes:

`mgos_sys_config_save(cfg, false, NULL);`

However i don't have the `cfg` pointer... Is there some easy way to do following?

```
mgos_sys_config_set_my_variable(123);
mgos_sys_config_store_all_variables_to_persistent_flash_memory();
```

rojer, 2020-01-10:
use &mgos_sys_config as the pointer, it's the global config instance.

Harvie, 2020-01-10:
> use &mgos_sys_config as the pointer, it's the global config instance.

Works a treat! THANKS!

Harvie, 2020-01-10:
Nice. Can it store higher values than "d" 4294967295? Eg. uint64_t ?

rojer, 2020-01-10:
no, it directly corresponds to C's "unsigned int" type which is 32 bit wide on all our platforms.
adding 64 bit types would be a separate change, which will have to wait.

Harvie, 2020-01-11:
I have another problem with the "d" type. :-/

It builds for me on Linux (64b build of mos tool 2.16.0), but i've commited the mos.yml to git and my friend tries to compile it on windows and gets following error:

`Error: json: error calling MarshalJSON for type build.ConfigSchemaItem: invalid schema value: 4294967295 (type: int64)`

I suspect the problem might lay in fact that windows build provided on website:
https://mongoose-os.com/downloads/mos-release/win/mos.exe
is built as 32b binary `mos.exe: PE32 executable (console) Intel 80386, for MS Windows`
so it can't parse that number in mos.yml to generate json properly...

Can you please provide 64b build for windows?

rojer, 2020-01-11:
try 4294967295.0 instead

rojer, 2020-01-11:
but it's an interesting problem. i don't see why would 32 and 64-bit, windows and linux builds would behave differently here. in theory they should have the same YAML parser, which is in Go and is platform-agnostic. which mos version doe she use?

Harvie, 2020-01-11:
He used some mos tool with version denoted by date 201912* (not sure exactly), so i made him to get the same 2.16.0 version as i use on linux and the problem persisted. Docker image is 3.3-r2 for both of us, but i guess this does not matter as YAML is probably preprocessed before entering docker build.

rojer, 2020-01-11:
i'll try to find a win32 virtual machine to test

Harvie, 2020-01-11:
I don't think you need win32 machine... 32b binaries will run on 64b system... At least on Linux it works that way. Not sure about windows.

rojer, 2020-01-11:
having trouble cross-compiling for 32b linux... anyway, can you file a new issue for this? i'll take a look later.