cesanta/mongoose-os #527: Removing files from the filesystem

dimitrov-adrian, 2020-01-22:
Is there a way to remove specific files that comes from libraries, from the filesystem. I found https://github.com/mongoose-os-libs/mjs/issues/5, but it seems to doesn't works at all.

For example, I would like to completely remove `polyfill.js`, `api_i2c.js`, `api_bitbang.js` and etc.

rojer, 2020-01-31:
currently there is no way to do that.
the list is actually passed from mos utility during build - it performs expansion of passes the list down.
a flag could be added to perform filtering to exclude files.

kzyapkov, 2020-09-30:
Recent versions of `mos` support `fs_filters` which can be used to control contents of the filesystem, ex:

```fs_filters:
  - include: "index.html*"
  - include: "init.js"
  - include: "api_math.js"
  - include: "api_events.js"
  - include: "api_http.js"
  - include: "api_net.js"
  - include: "api_timer.js"
  - include: "api_rpc.js"
  - include: "api_esp32.js"
  - include: "api_file.js"

  - exclude: "api_*"
  - exclude: "polyfill.js"
```

closed 2020-09-30 by kzyapkov