cesanta/mongoose-os #528: WDT resets on ESP8266 if mgos_gpio_enable_int is not called

zdila, 2020-01-22:
On ESP8266 if `mgos_gpio_set_int_handler(4, MGOS_GPIO_INT_EDGE_ANY, handler, NULL);` is called without calling `mgos_gpio_enable_int(4);` then after making a pulse on GPIO4 system freezes and WDT is triggered. After that WDT reset it freezes on boot and loops forever: https://pastebin.com/6RdT0txh

To prevent freeze on following boot there is an workaround:
```c
void mgos_app_preinit(void) {
   mgos_gpio_enable_int(4);
}
```

Calling `mgos_gpio_set_int_handler `without `mgos_gpio_enable_int` has probably no sense but I think it should not freeze the system.

pkjq, 2020-05-24:
I have the same issue with disabled interrupt on any pin. I create pin interrupt handler and enable interrupt for it. When he triggered i disable interrupt for pin for any time and then set timer. When timer is fired i enable interrupt for this pin. Some times after disable pin interrupt mcu is hang: wifi connection lost and etc. If i comment code with disabling interrupt for pin and use boolean variable for filtering interrupts mcu working correctly. Problem reporducing if many interrupts triggered when it's disabled.

pkjq, 2020-05-24:
I was made some experiments on ESP8266 and i have next theory:
In code of int-handler exceutes:
```c
GPIO_REG_WRITE(GPIO_STATUS_W1TC_ADDRESS, 1 << pin);
```
by calls function `mgos_gpio_hal_clear_int`. [see](https://github.com/cesanta/mongoose-os/blob/3948c6b2495212f2e46fefc9006209d5acc3eea6/src/mgos_gpio.c#L82). But if we disable interrupt in it's handler race-condition may be occured and nobody clear interrupt bit. I think you must call `mgos_gpio_hal_clear_int` too when interrupt of current pin is disabled.

rojer, 2020-05-24:
yes, i think you're right. thank you!

closed 2020-05-24 by rojer

pkjq, 2020-05-25:
@rojer , maybe it's actual and for esp32 too.