cesanta/mongoose-os #529: Compilation failed: mgos_sys_config_get_debug_event_level undefined function

pkjq, 2020-02-02:
```/mongoose-os/src/mgos_debug.c: In function 'mgos_debug_write':
/mongoose-os/src/mgos_debug.c:103:3: error: implicit declaration of function 'mgos_sys_config_get_debug_event_level' [-Werror=implicit-function-declaration]
   if (cs_log_cur_msg_level <= mgos_sys_config_get_debug_event_level())
   ^
```

rojer, 2020-02-02:
debug.event_level is defined in the core lib, in [this commit](https://github.com/mongoose-os-libs/core/commit/c7359ef09e0f75bd693d88672ba9665b84d0e46a).
make sure your core lib is up to date.

closed 2020-02-02 by rojer

pkjq, 2020-02-02:
```
...
i2c: Does not exist, cloning from "https://github.com/mongoose-os-libs/i2c"...
i2c: Checking out 2.16.0...
i2c: Done, hash 4bf5e876983b1767e50eab2bb4a71f4b225520e9
i2c: Hash is updated:  -> 4bf5e876983b1767e50eab2bb4a71f4b225520e9
i2c: Prepared local dir: "/data/fwbuild-volumes/2.16.0/apps/xxx/esp8266/build_contexts/build_ctx_790052265/deps/i2c"
Handling lib "vfs-dev-part"...
Handling lib "vfs-fs-spiffs"...
vfs-dev-part: Does not exist, cloning from "https://github.com/mongoose-os-libs/vfs-dev-part"...
vfs-fs-spiffs: Does not exist, cloning from "https://github.com/mongoose-os-libs/vfs-fs-spiffs"...
vfs-dev-part: Checking out 2.16.0...
vfs-dev-part: Done, hash 94426df449f807e4268e1d1a3ee5146da85902da
vfs-dev-part: Hash is updated:  -> 94426df449f807e4268e1d1a3ee5146da85902da
vfs-dev-part: Prepared local dir: "/data/fwbuild-volumes/2.16.0/apps/xxx/esp8266/build_contexts/build_ctx_790052265/deps/vfs-dev-part"
vfs-fs-spiffs: Checking out 2.16.0...
vfs-fs-spiffs: Done, hash 6fef5f74fc5f8e07b6324759066c0e9d06653693
vfs-fs-spiffs: Hash is updated:  -> 6fef5f74fc5f8e07b6324759066c0e9d06653693
vfs-fs-spiffs: Prepared local dir: "/data/fwbuild-volumes/2.16.0/apps/xxx/esp8266/build_contexts/build_ctx_790052265/deps/vfs-fs-spiffs"
The flag --repo is not given, going to use mongoose-os repository
mongoose-os: Does not exist, cloning from "https://github.com/cesanta/mongoose-os"...
mongoose-os: Checking out master...
mongoose-os: Done, hash c31a7453f6754f80acab6b1b229fc097bf411f37
...
```
...checkout master. But mongoose-os-libs/core uses 2.16.0.
What's wrong in my config?

rojer, 2020-02-02:
the version of mongoose-os app uses is determined by `mongoose_os_version` in app's manifest.
what is it set to?

pkjq, 2020-02-02:
@rojer , thanks. It's was be defined as 'master'. It's my mistake :-)