cesanta/mongoose-os #531: Support of HLW8012 energy meter

ellavas, 2020-03-20:
Hello,

Many devices (like Shelly 1PM or SonOff Pow R2) are using the HLW8012 sensor to monitor the power consumption. 
Would it be possible to integrate the HLW8012 library (https://www.arduinolibraries.info/libraries/hlw8012) to Mongoose OS ?

Thanks

pimvanpelt, 2020-03-20:
I had it on my todo list (because I have those exact Shelly 1PM units), but
I'm afraid I won't be able to get to it in the short term. I can work on
this in ~June at the earliest, although it would be pretty straight forward
to add because the chip is a straight forward I2C speaker. By the way,
Shelly 2.5 uses ADE7953 which would've been the first one on my list to
implement a driver for.

ellavas, 2020-03-20:
Shelly firmwares are based on Mongoose, so it should not be too difficult to integrate it.

albanlud, 2020-03-20:
That would be awesome to have this power monitoring available for Shelly 1 PM !
I can't wait !

pimvanpelt, 2020-03-27:
I just submitted https://github.com/mongoose-os-libs/ade7953-i2c which is for Shelly 2.5 and confirmed working on my unit:
```
esp8266_10C60B 5329 1585312791.954 2|mgos_ade7953.c:105      val=4475 hertz=49.988831
esp8266_10C60B 5330 1585312791.961 2|mgos_ade7953.c:114      val=6101779 volts=234.683807
esp8266_10C60B 5331 1585312791.967 2|mgos_ade7953.c:130      reg=794 val=1773 amperes=0.017730
esp8266_10C60B 5332 1585312791.973 2|mgos_ade7953.c:130      reg=795 val=46998 amperes=0.469980
esp8266_10C60B 5333 1585312791.982 2|main.c:16               frequency=49.988831 voltage=234.683807 currentA=0.017730 currentB=0.469980
```

I do not currently have access to a spare Shelly 1PM for the HLW8012 driver. I've ordered a few units to be able to write and test the driver, and as soon as they arrive I'll work on that next.

rojer, 2020-03-27:
this is great, i can integrate it into https://github.com/mongoose-os-apps/shelly-homekit now

pimvanpelt, 2020-03-27:
By the way, Shelly 1PM seems to be based on BL 0937, not HLW8012, but they are often referenced together so likely the same interface. I will update this issue once I have my hands on either, but could probably use some help testing that one.

rojer, 2020-03-27:
i only have 1 and 2.5 myself, unfortunately

albanlud, 2020-03-27:
I got a "ready to be tested" shelly 1PM, so please do not hesitate to ask for help !

ellavas, 2020-03-29:
> I got a "ready to be tested" shelly 1PM, so please do not hesitate to ask for help !

Could you check which chip is soldered in the Shelly 1PM ?

albanlud, 2020-03-30:
It is a BL 0937 ...

ellavas, 2020-04-28:
I’ve ported the Arduino HLW8012 power meter sensor library.

It's working fine with the Shelly 1PM. (BL0937)
You can have a look here:

https://github.com/ellavas/mongoose-lib-hlw8012

albanlud, 2020-04-28:
Great job !!

rojer, 2020-04-28:
interesting, i need to take a look. anychance you can integrate it into the firmware as well? i don't even have 1PM to test.