cesanta/mongoose-os #532: `mgos_clear_timer()` is broken on Ubuntu

pimvanpelt, 2020-03-29:
# Reproduction
```
mgos_timer_id id;
id = mgos_set_timer(1000, true, cb, NULL);
mgos_clear_timer(id);
```

# Behavior

Expected: the callback `cb` is no longer called.
Witnessed: the callback `cb` is called every 1000ms.

# Explanation
`mgos_clear_timer()` uses a platform dependent hack to signal if the passed timer `id` is a hardware or software timer. Under the covers, `mgos_timer_id` is of type `uintptr_t` and it points to different structures depending on the timer type (hw || sw).
On Ubuntu platform, the pointer to the struct cannot be masked by `MGOS_SW_TIMER_MASK` upon creation, and thus it is always considered a hardware timer. Then, `mgos_clear_timer()` will call `mgos_clear_hw_timer()` always, while it should be calling `mgos_clear_sw_timer()`. Hardware timers are not implemented in Ubuntu (and are unlikely to ever be implemented).

# Workaround
As a workaround, I have rewritten `mgos_clear_timer()` such that it is aware of the platform:
```
IRAM void mgos_clear_timer(mgos_timer_id id) {
#if PLATFORM == ubuntu
  if (id == MGOS_INVALID_TIMER_ID) {
    return;
  } else {
    mgos_clear_sw_timer(id);
  }
#else
  if (id == MGOS_INVALID_TIMER_ID) {
    return;
  } else if (id & MGOS_SW_TIMER_MASK) {
    mgos_clear_sw_timer(id);
  } else {
    mgos_clear_hw_timer(id);
  }
#endif
}
```

I spent a few hours looking for a better solution than this stop-gap, but due to the fact that the `mgos_timer_id` pointer is overloaded in the pointer value itself, and moreover points to two different types of struct depending on the hardware/software timer, I am drawing a blank on a more elegant solution that won't turn into a huge refactoring. PTAL if the #if/#endif here is good enough.

rojer, 2020-03-29:
yes, i think we can assume all timers are software timers on Ubuntu.
we can set MGOS_SW_TIMER_MASK above to ((uintptr_t) -1) so the condition always matches.

pimvanpelt, 2020-03-29:
Ah, I see. Yes that is a pretty good solution; Can you submit that? Saves me a PR :)

rojer, 2020-03-29:
done

closed 2020-03-29 by rojer