cesanta/mongoose-os #535: mgos_bitbang support clock signal?

Harvie, 2020-04-08:
Hello,
can you please add possibility to output clock signal on second GPIO using `mgos_bitbang` ?

rojer, 2020-04-10:
mgos_bitbang is not currently suited for outputting continuous signals. you are better off using hardware timer.

Harvie, 2020-04-16:
what does "continuous signal" mean? My use case is not exactly streaming a lot of data non-stop. Just sending few bytes now and then. Just short commands to configure external IC...

rojer, 2020-04-16:
ok, so you need to drive two pins - clock and then bitbang data, correct?
i see how it might be useful, but making it generic is a bit problematic. you'll need to specify whether to change data on rising or falling edge of the clock, etc.
i'd say, you'll need to build your own solution, if more people come along and ask for it, we can think about making it a library feature.