cesanta/mongoose-os #537: Add mgos_invoke_cb() option to use xQueueSendToFront...()

Harvie, 2020-04-21:
I need to use relatively complex code in ESP32 GPIO ISR (to handle complex protocol in time).
But it turned out, i can't just tag everything IRAM_ATTR... It panics and i don't know why.

What is working for me is using mgos_invoke_cb to switch from ISR context to main context, where i can use complex libraries... But i feel like it's slow and other events and timers can mess me up, so i can't handle the communicaton on time. Is there way around this?

`mgos_invoke_cb()` currently uses `xQueueSendToBackFromISR()`. What if we used `xQueueSendToFrontFromISR()` instead? That way i probably can get more priority over other stuff to be able to handle my communication on time... And get executed ASAP after return from ISR...

https://www.freertos.org/xQueueSendToFront.html

```C
IRAM bool mgos_invoke_cb(mgos_cb_t cb, void *arg, bool from_isr) {
  struct mgos_event e = {.cb = cb, .arg = arg};
  if (from_isr) {
    BaseType_t should_yield = false;
    if (!xQueueSendToBackFromISR(s_main_queue, &e, &should_yield)) {
      return false;
    }
    YIELD_FROM_ISR(should_yield);
    return true;
  } else {
    return xQueueSendToBack(s_main_queue, &e, 10);
  }
}
```

rojer, 2020-04-21:
we can add a variant that sends to front instead - `mgos_invoke_cb_front`, perhaps.
feel free to send a PR.

Harvie, 2020-04-21:
I can do the PR, but i am not FreeRTOS expert... Can you please confirm this will actualy work in the way i expect?

So my callback will get executed immediately instead of waiting for other jobs that might have been queued before that?

rojer, 2020-04-21:
you can add elements to queue front, yes.
it will not get executed immediately, but it will be executed as soon as previosu callback currently executing on the main task finished executing.
how about you try it, see if it makes a difference for you and if it does, send a PR.
i don't want to add something that is not used, so far i have not had the need to do it.

Harvie, 2020-04-21:
Sure. I will have to first check what causes more trouble in my case. Events already being executed or the events that are queued.

BTW is there something else that you would reccomend to tackle my issue with having to do complex communication in ISR? I was thinking about creating new RTOS task in ISR to handle the communication. That way the slow/blocking mgos event might get preempted and my code might run fast enough. But i am not sure about how fast/realtime this would be...

Harvie, 2020-04-21:
Is there way i can check if there's something waiting in mgos queue? So i can use it in ISR to check if that's the case?

rojer, 2020-04-21:
creating a task is possible and might be warranted in your case. if you create it with priority higher than  MGOS_TASK_PRIORITY (e.g. MGOS_TASK_PRIORITY+1), it will preempt mgos if you send a queue item to it. just be careful invoking mgos APIs from it, as mgos is not thread-safe. you'll need to use mgos_invoke_cb from your task to interact with mgos.

Harvie, 2020-04-21:
> mgos is not thread-safe. you'll need to use mgos_invoke_cb from your task to interact with mgos.

I probably only need the following calls to mgos. Is there some reason why it should cause problems?

`mgos_uptime_micros(); mgos_ints_disable(); mgos_msleep(); and mgos_gpio_*`

rojer, 2020-04-21:
no, those should be fine.

Harvie, 2020-04-21:
Maybe i can create high priority RTOS task which would run similar event queue/loop to what mgos uses, but i will only use it for realtime stuff. Maybe this idea might be even generalized and implemented directly in mongoose as some kind of "multiqueue" support.

rojer, 2020-04-21:
> Maybe i can create high priority RTOS task which would run similar event queue/loop to what mgos uses, but i will only use it for realtime stuff

yes, that's exactly what i'm suggesting.
let's hold off generalizing for now.

Harvie, 2020-10-03:
> let's hold off generalizing for now.

Too baaad, i've already generalized the heck out of it :-) I've came up with relatively elegant and simple120 LoC solution.

It looks like this from user perspective:

```C
  //Configure some basic parameters
  pq_handle pqh;
    pq_set_defaults(&pqh);
    pqh.idle_after_ms = 6000;
    pqh.idle_cb = idle_cb;
    pqh.idle_cb_arg = "IDLE ARG";
    pqh.prio = MGOS_TASK_PRIORITY+1;

  //Create new FreeRTOS task running new FreeRTOS queue based event loop very similar to mongoose os
  pq_start(&pqh);

  //Invoke callback same way the mgos_invoke_cb() does
  pq_invoke_cb(&pqh, pq_test_cb, NULL, (void *)"CB ARG", false, false);
```

Benefits when compared to original mongoose event loop:
  * Can have multiple event loops with various priorities and yes, they will preempt each other even when callback is still running
  * Can use invoke_cb() parameter to do xQueueSend "to FRONT" instead of "to BACK" for task with extreme priority to get even more priority than the other events in high-priority task (use this responsibly!)
  * If needed i can reimplement original mongoose event loop API using few single line #define macros.
  * Each queue can optionaly register IDLE callback with custom timeout. So when there was no event in the queue for defined time period, it will automaticaly run the idle callback to do some "housekeeping" tasks with extremely low priority.
  * I am willing to provide the code once it will get sufficiently battle-tested.
  * Currently the code is provided in form of mongoose module, but can be easily integrated to mongoose core to work as both API for 3rd party modules and replacement for current mongoose os event loop.

Harvie, 2020-10-30:
I would need to move few mongoose defines (eg. MGOS_TASK_PRIORITY) to header files so i don't have to redefine them in my parallel event loop module, to keep the code compatible with upstream changes. Would you mind looking at it? #554

Harvie, 2020-11-09:
So here is the parallel event loop module i've been talking about:

https://github.com/Harvie/pq
https://github.com/mongoose-os-libs/freertos/issues/3

Harvie, 2021-03-18:
What do you think about that pq project?
Would you accept PR that would integrate it to mongoose os to replace currently used event loop without breaking current API?