cesanta/mongoose-os #541: Automatic power saving esp32 when idle.

Harvie, 2020-05-31:
Hello, i am using mongoose os on ESP32 based power sensitive project.

Is there any chance to change mongoose behaviour in such way that it would automaticaly bring esp32 to some kind of low-power (sleep or low CPU clock frequency) mode until when there is not event or interrupt handled and automaticaly turn the power back on, once something starts to happen again...

Kinda similar to how my laptop increases power when i start compiling of mongoose project and later when it's done, the everything gets cool again and fan turns off.

Our device has temperature sensor to measure ambient temperature, but it gets affected by heat from esp32. We tried to address this issue in mechanical design, but it is still thrown off by couple of degrees C.

DrBomb, 2020-05-31:
All the esp-idf methods are available when you compile for the esp32. Here's some pages that might do what you need:

[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html)
[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/power_management.html)

Of course, none of these power modes are supported by mongoose. Feel free to let us know your results.

Harvie, 2020-05-31:
I will try the following to see if that will work... Hopefully it will be helpfull without any further code modifications... But we can't really tell unless we test this.

```yaml
build_vars:
  ESP_IDF_SDKCONFIG_OPTS: >
    ${build_vars.ESP_IDF_SDKCONFIG_OPTS}
    CONFIG_PM_ENABLE=y
    CONFIG_PM_DFS_INIT_AUTO=y
```

Harvie, 2020-10-30:
Turns out that configuration in previous post actualy does something. Because it downscaled CPU clock to such extent that it actualy broken the UART timing during low-load scenarios. Which means it works! Also @rojer provided his expertise and fixed the UART problem by reconfiguring it to use independent clock source ( #556 ), so thanks to him, we can now have both power saving mode and reliable UART at the same time. yay!

kargarisaac, 2022-02-27:
I have the same goal. I use deep sleep mode and wake it up every hour to measure sth and send data via mqtt and again go to deep sleep mode. I wonder if there is anything else I can do to use less power. Should I go for arduino to have this capability or adding the above configs to `mos.yml` does the job?