cesanta/mongoose-os #545: http_server for esp32 core dumps with self-signed certificates

nliviu, 2020-07-31:
Configure [http_server](https://github.com/mongoose-os-libs/http-server) for esp32 with self-signed certificates.
`https://IP`core dumps in Chrome and Opera for Windows and MacOS. It works with Firefox for Windows and MacOS, also with Internet Exlorer and Edge for Windows.

Chrome
```
[Jul 30 11:00:30.833] mg_ssl_if_mbedtls.c:31  0x3ffc639c ciphersuite: TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
[Jul 30 11:00:30.837] SW ECDH curve 3
[Jul 30 11:00:32.043] mg_ssl_if_mbedtls.c:31  0x3ffc639c is a fatal alert message (msg 46)
[Jul 30 11:00:32.052] mg_ssl_if_mbedtls.c:31  0x3ffc639c mbedtls_ssl_handle_message_type() returned -30592 (-0x7780)
[Jul 30 11:00:32.059] mg_ssl_if_mbedtls.c:31  0x3ffc639c mbedtls_ssl_read_record() returned -30592 (-0x7780)
[Jul 30 11:00:32.065] mg_ssl_if_mbedtls.c:207 0x3ffc6998 mbedTLS error: -0x7780    <- 0x3ffc6998=struct mg_connection *nc
[Jul 30 11:00:32.074] mg_ssl_if_mbedtls.c:31  0x3ffc639c ciphersuite: TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
[Jul 30 11:00:32.077] SW ECDH curve 3
[Jul 30 11:00:32.828] mg_ssl_if_mbedtls.c:31  0x3ffc639c is a fatal alert message (msg 46)
[Jul 30 11:00:32.836] mg_ssl_if_mbedtls.c:31  0x3ffc639c mbedtls_ssl_handle_message_type() returned -30592 (-0x7780)
[Jul 30 11:00:32.844] mg_ssl_if_mbedtls.c:31  0x3ffc639c mbedtls_ssl_read_record() returned -30592 (-0x7780)
[Jul 30 11:00:32.849] mg_ssl_if_mbedtls.c:207 0x3ffc831c mbedTLS error: -0x7780    <- 0x3ffc831c=struct mg_connection *nc
[Jul 30 11:00:32.858] mg_ssl_if_mbedtls.c:31  0x3ffc639c ciphersuite: TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
[Jul 30 11:00:32.862] SW ECDH curve 3
[Jul 30 11:00:33.613] mg_ssl_if_mbedtls.c:31  0x3ffc639c is a fatal alert message (msg 46)
[Jul 30 11:00:33.622] mg_ssl_if_mbedtls.c:31  0x3ffc639c mbedtls_ssl_handle_message_type() returned -30592 (-0x7780)
[Jul 30 11:00:33.630] mg_ssl_if_mbedtls.c:31  0x3ffc639c mbedtls_ssl_read_record() returned -30592 (-0x7780)
[Jul 30 11:00:33.635] mg_ssl_if_mbedtls.c:207 0x3ffc9c80 mbedTLS error: -0x7780    <- 0x3ffc9c80=struct mg_connection *nc
[Jul 30 11:00:33.680] CORRUPT HEAP: multi_heap.c:477 detected at 0x3ffc6c7c
```

```
Loaded core dump from last snippet in  /core
0x40087e9f in invoke_abort ()
    at /opt/Espressif/esp-idf/components/esp32/panic.c:156
156             *((int *) 0) = 0;
#0  0x40087e9f in invoke_abort ()
    at /opt/Espressif/esp-idf/components/esp32/panic.c:156
#1  0x40088024 in abort ()
    at /opt/Espressif/esp-idf/components/esp32/panic.c:171
#2  0x40091f1a in multi_heap_assert (condition=false, line=477,
    address=1073507452,
    format=0x3ffb2e91 "CORRUPT HEAP: multi_heap.c:%d detected at 0x%08x\n")
    at /opt/Espressif/esp-idf/components/heap/multi_heap_platform.h:54
#3  0x4009233c in multi_heap_free_impl (heap=0x3ffbb320, p=0x3ffc6c80)
    at /opt/Espressif/esp-idf/components/heap/multi_heap.c:477
#4  0x40081d34 in heap_caps_free (ptr=<optimized out>)
    at /opt/Espressif/esp-idf/components/heap/heap_caps.c:272
#5  0x4008225c in _free_r (r=0x3ffc08b0, ptr=0x3ffc6c80)
    at /opt/Espressif/esp-idf/components/newlib/syscalls.c:42
#6  0x4000beca in ?? ()
#7  0x4012fc3f in mem_free (rmem=0x3ffc6c80)
    at /opt/Espressif/esp-idf/components/lwip/lwip/src/core/mem.c:176
#8  0x4012fcca in do_memp_free_pool (mem=0x3ffc6c80,
    desc=0x3f415e78 <memp_TCP_PCB>)
    at /opt/Espressif/esp-idf/components/lwip/lwip/src/core/memp.c:432
#9  0x4012fd1e in memp_free (type=MEMP_TCP_PCB, mem=0x3ffc6c80)
    at /opt/Espressif/esp-idf/components/lwip/lwip/src/core/memp.c:489
#10 0x40130968 in tcp_close_shutdown (pcb=0x3ffc6c80,
    rst_on_unacked_data=<optimized out>)
    at /opt/Espressif/esp-idf/components/lwip/lwip/src/core/tcp.c:290
#11 0x40130a1e in tcp_close (pcb=0x3ffc6c80)
    at /opt/Espressif/esp-idf/components/lwip/lwip/src/core/tcp.c:409
#12 0x4016b0bc in tcp_close_tcpip (arg=0x3ffc6c80)
    at common/platforms/lwip/mg_lwip_net_if.c:357
#13 0x4016b09f in xxx_tcpip (arg=0x3ffb5d00 <mgos_task_stack+7752>)
    at common/platforms/lwip/mg_lwip_net_if.c:70
#14 0x4012f9d1 in tcpip_thread (arg=<optimized out>)
    at /opt/Espressif/esp-idf/components/lwip/lwip/src/api/tcpip.c:101
```

Firefox waits with "Warning: Potential Security Risk Ahead"
```
[Jul 30 11:03:19.335] mg_ssl_if_mbedtls.c:31  0x3ffc639c ciphersuite: TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
[Jul 30 11:03:19.339] SW ECDH curve 3
[Jul 30 11:03:20.550] mg_ssl_if_mbedtls.c:31  0x3ffc639c is a fatal alert message (msg 48)
[Jul 30 11:03:20.559] mg_ssl_if_mbedtls.c:31  0x3ffc639c mbedtls_ssl_handle_message_type() returned -30592 (-0x7780)
[Jul 30 11:03:20.566] mg_ssl_if_mbedtls.c:31  0x3ffc639c mbedtls_ssl_read_record() returned -30592 (-0x7780)
[Jul 30 11:03:20.571] mg_ssl_if_mbedtls.c:207 0x3ffc84ac mbedTLS error: -0x7780
[Jul 30 11:03:20.577] mgos_mongoose.c:66      New heap free LWM: 214596
```

After accepting the exception
```
[Jul 30 11:04:55.962] mg_ssl_if_mbedtls.c:31  0x3ffc639c ciphersuite: TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
[Jul 30 11:04:55.966] SW ECDH curve 3
[Jul 30 11:04:56.715] mgos_mongoose.c:66      New heap free LWM: 214012
[Jul 30 11:04:56.944] mgos_http_server.c:180  0x3ffc6984 HTTP connection from 192.168.0.99:3796
[Jul 30 11:04:56.949] mgos_http_server.c:188  0x3ffc6984 GET /
[Jul 30 11:05:01.388] main.c:25               Tick uptime: 140.96, free_heap: 226708, min_free_heap: 214012
```

fcolasante, 2020-12-02:
Quote! Same for me.

rojer, 2020-12-19:
ok, i found the problem.
it has nothing to do with SSL as such, but rather with chromium's connection behavior when connecting to sites with untrusted certificates.
it establishes two connections in rapid succession: first fails with alert 46 (unknown certificate), second is successful. it continues to behave this way even after user has clicked through the "yes, i want to connect" page.
this looks like a minor bug in chromium - why create a connection, drop, then connect again?
anyway, even that is not the root cause. the root cause is a race to close the connection between client and server: client sends fatal TLS error, so server marks connection as MG_F_CLOSE_IMMEDIATELY. meanwhile, client also aborts the connection, in a way that sends back RST and causes LWIP to raise error signal, that immediately deallocates everything associated with the connection. the bug was that mongoose was then trying to free it again, causing a double free and corrupting heap.
i sent a fix out for review and it should land soon.

closed 2020-12-20 by cpq