cesanta/mongoose-os #546: How do i force mos tool to use my fork of mongoose-os repo?

Harvie, 2020-07-31:
Hello,
i've made fork of this repo. how do i specify in mos.yml that i want to use my fork for the build? so i can test the code before doing PR?

For libraries it is easy, just put it in deps in mos.yml, but i don't know how to do it with the mongoose itself...

eg.: https://github.com/harvie/mongoose-os/

DrBomb, 2020-07-31:
Hello harvie. You can use the `--repo` flag to point to a local mongoose-os repo on your local machine. That's the only way I know about as the mongoose-os module is managed by the build process.

Feel free to join us in the gitter chat as to avoid opening issues that aren't actual issues. Have a great day.

[https://gitter.im/cesanta/mongoose-os](https://gitter.im/cesanta/mongoose-os)

closed 2020-07-31 by DrBomb

Harvie, 2020-07-31:
This seems to work. Thank you!