cesanta/mongoose-os #548: CVE-2020-12638 // WPA2 downgrade in Espressif

NS-K, 2020-08-03:
Affected Devices : ESP32 & ESP8266

Espressif's Security Advisory :
https://www.espressif.com/sites/default/files/advisory_downloads/Security%20advisory_%20authentication%20bypass.pdf

Proof of Concept :
https://lbsfilm.at/blog/wpa2-authenticationmode-downgrade-in-espressif-microprocessors

rojer, 2020-08-03:
thanks for reporting. this is indeed a serious issue. i will update ESP32 SDK soon but for ESP8266 it's not that easy: the fix is only made available for RTOS SDK and NON-OS SDK 3.0+ while we are still using v2.2.
i filed https://github.com/espressif/ESP8266_NONOS_SDK/issues/326, let's see what they have to say.

d-a-v, 2020-08-03:
It is fixed in esp8266 Arduino 2.7.4 and master with a workaround confirmed to work by the author.

rojer, 2020-08-03:
i see, https://github.com/esp8266/Arduino/pull/7486
we should be able to do something like that too.

NS-K, 2020-08-03:
It seems to be very easy to implement on 2.X.X

rojer, 2020-08-04:
updated ESP-IDF and applied workaround for ESP8266

closed 2020-08-04 by rojer