cesanta/mongoose-os #551: Fail when calling mgos_ints_enable()

VaclavVejrosta, 2020-09-03:
Hi,
this problem occures when I enable interrupts with function `mgos_ints_enable()` . But it never happened with mos version 2.17. It fails only with mos version 2.18. It fails always when this function is called. Please see attached logs:

[image]

[image]

VaclavVejrosta, 2020-09-03:
Problem solved in issue #550

closed 2020-09-03 by VaclavVejrosta

rojer, 2020-09-03:
looks like it's related to multicore operation.
if you can create a simple app that reproduces it, i can take a look.