cesanta/mongoose-os #555: C++ undefined references? Not linking like it should do.

Respawnerhd, 2020-10-15:
Following errors iam getting:

```
undefined reference to `mg
undefined reference to `mg_mgr_poll'_bind'
undefined reference to `mg_mgr_init'
undefined reference to `mbuf_remove'
```

how do i fix this?
If i execute this: `g++ main.cpp -lmongoose `
it gives me that:
`c:/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/bin/ld.exe: cannot find -lmongoose`

DrBomb, 2020-10-15:
Are you doing a local or a remote build? You don't have to execute g++ at any moment to build a mongoose-os app.

Please read the docs [https://mongoose-os.com/docs/mongoose-os/userguide/build.md](https://mongoose-os.com/docs/mongoose-os/userguide/build.md)

Respawnerhd, 2020-10-16:
> Are you doing a local or a remote build? You don't have to execute g++ at any moment to build a mongoose-os app.
> 
> Please read the docs https://mongoose-os.com/docs/mongoose-os/userguide/build.md

Iam doing it locally on my pc with visual studio code without any docker. 

In theyre docs states that i just need to place the mongoose.c and mongoose.h in my build directory

DrBomb, 2020-10-18:
Looks like this issue is related to "mongoose" the networking library instead of "mongoose-os" which is the IoT development framework. This is the correct repo to fill the issue [https://github.com/cesanta/mongoose](https://github.com/cesanta/mongoose)

closed 2020-10-18 by DrBomb