cesanta/mongoose-os #565: ESP-IDF 4.2 update

rojer, 2020-12-27:


nliviu, 2020-12-28:
It looks like there is a problem with I2C:
```
[Dec 28 21:37:42.284] esp32_main.c:65         ESP-IDF 4.2-r1
...
[Dec 28 21:37:43.635] esp32_i2c_master.c:402  I2C0 init ok (SDA: 21, SCL: 22, freq: 100000)
...
[Dec 28 21:37:44.228] mgos_bh1750.c:150       Could not set mtime
[Dec 28 21:37:44.232] mgos_bh1750.c:99        Failed to init BH1750 @ 0x23
```
The error is issued after a `mgos_i2c_write`.
https://github.com/mongoose-os-libs/bh1750/blob/f77905d448ebcfe2eea4e3fb4ebf66d1e21b4e70/src/mgos_bh1750.c#L149-L152

rojer, 2020-12-28:
@nliviu thanks for reporting, fixed by https://github.com/mongoose-os-libs/i2c/commit/d01074bf3c125449d87c508ea1a5bfa27c03c9d9

kzyapkov, 2020-12-29:
`ESP32-PICO-D2` needs `CONFIG_FLASHMODE_DIO=y` appended to `ESP_IDF_SDKCONFIG_OPTS`, specifying the mode in `--esp-flash-params` isn't enough. This was not the case with IDF v3.2.