cesanta/mongoose-os #566: call to IDF function gpio_install_isr_service() fails with error code 0x105

scaprile, 2021-01-06:
A call to gpio_install_isr_service() always fails with error code 0x105:
ESP_ERR_NOT_FOUND (0x105): Requested resource not found (No free interrupt found with the specified flags)
even on an empty project (attached)

Is this expected behavior ? (Does mOS use gpio_isr_register() ?)
Function header reads:

> esp_err_t gpio_install_isr_service(int intr_alloc_flags)
>     Install the driver’s GPIO ISR handler service, which allows per-pin GPIO interrupt handlers.
>     This function is incompatible with gpio_isr_register() - if that function is used, a single global ISR is registered for all GPIO interrupts. If this function is used, the ISR service provides a global GPIO ISR and individual pin handlers are registered via the gpio_isr_handler_add() function.
> 

I'm on stable 2.18.0
Regards

[isr-service.zip](https://github.com/cesanta/mongoose-os/files/5777653/isr-service.zip)

gibzwein, 2021-02-14:
Hi,
have You solved that issue?
Regards

scaprile, 2021-02-14:
> have You solved that issue?

Nope, I've let the project aside until I either get support or have the extra time to play around with this. I'll post it here either way, if ever.

gibzwein, 2021-02-14:
Thanks for reply

gadget-man, 2023-01-25:
Did you ever get to the bottom of this? I've seen a couple of issues of interrupts failing mid-code - only a reboot works (calling an RPC function to create a new button_handler doesn't work even though it returns success. I'm wondering whether it might be linked to this issue if `gpio_install_isr_service` isn't being used.

scaprile, 2023-01-25:
Hi gadget-man, no, I abandoned this and moved on.