cesanta/mongoose-os #569: Fail in softAP mode

VaclavVejrosta, 2021-01-25:
Hi,

since I have updated my mos to version 2.19 I can see a quite new bug. It happens during pairing of my "smart device". First of all the device creates an AP and I connect to it with my mobile. Then the device gets the data about the current wifi network and then it tries to connet to wifi.  But when disconnecting the softAP mode an error occures and the device is restarted. However, the device is connected to wifi correctly after reboot.

Please see log:
[Log.txt](https://github.com/cesanta/mongoose-os/files/5867106/Log.txt)

The problem never occured with mos 2.18.  By the way, how can I downgrade my mos back to 2.18? I used command `mos update version 2.18.0 ` but it fails.

DrBomb, 2021-01-25:
Hi, although I cannot help you with the actual issue. You can add [these lines](https://github.com/mongoose-os-apps/demo-js/blob/master/mos.yml#L6-L8) to your `mos.yml` to pin the build to a specific version. Put `2.18.0` instead of `${mos.version}`

rojer, 2021-01-26:
"Stack canary watchpoint triggered" - this suggests stack overflow, and that is a huge stack,indeed.
looks like you're sending logs to websocket, and that log entry is emitted from networking stack itself... i think the problem might be because between 2.18 and 2.19 we redirected IDF logging to the mos logging handler, and now it goes through all the other logging handlers.
try turning off fancy logging, does it still happen?

VaclavVejrosta, 2021-01-28:
I have disabled the ws logs and it helped. The problem disappeared. However I use the ws logs in my products so unfortunately this is not the solution for me.

rojer, 2021-01-28:
i see. i think the task mentioned is the timer task, can you increase CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH from 4096 to, say. 6144 and see if that helps?

VaclavVejrosta, 2021-01-29:
I tried to increase the value up to 16 384 and it didn't help.

VaclavVejrosta, 2021-02-08:
Have you got any other ideas?

rojer, 2021-02-08:
i might, let me think about it some more...