cesanta/mongoose-os #574: STM32F1 support

Harvie, 2021-03-13:
Mongoose website states support for STM32 F4, L4, F7.
Is there any chance of supporting popular STM32F1?

I've noticed that @cpq have done some reseach on this in 2018...

rojer, 2021-03-13:
sorry, no chance of F1 support, it's too underpowered. F2 is ok.

Harvie, 2021-03-13:
OK, thanks for info.

BTW what is lacking about F1? Mongoose seems to have very little overhead because it is based on FreeRTOS. Should run on random potato without major problems. Anyway i am in process of designing custom STM32 board. This is good, because i can easily swap F1 for F4 at this stage of development to get proper Mongoose support.

Also how do i specify whether `mos flash` should use serial bootloader or stlink SWD to upload firmware? I am not able to find any details on the mos tool flashing process for STM32. Therefore i am not sure which programming interfaces i should expose on my board. SWD seems more robust, allows debugging and does not need jumper to enable bootloader. But flashing using serial bootloader can be achieved without having stlink hardware. Is it possible to use both of these options with mos tool? Is this docummented somewhere?

I guess i can run `stm32flash` or `st-flash` manualy depending on chosen method. But it would be nice to have my board compatible with whatever mos tool can use natively.

cpq, 2021-03-15:
@Harvie an alternative is to use something like https://vcon.io/docs/#vcon--stm32-bluepill