cesanta/mongoose-os #577: Log spam with Shelly 2.5

selmaohneh, 2021-05-19:
I use a shelly 2.5 for my canvas. It seems to have some strange issues, so I activated its debug logging. This line spams into the log, all the time:

`2478235876 mgos_mqtt_conn.c:578    MQTT0 queue overflow!`

I don't get any answers in the official shelly forum, so I will try it here, since the log seems to originate from mongoose-os.

Do you have any additional info on how that log can be prevent or whats the reasvon for that?

kzyapkov, 2021-05-19:
This is an issue with the Shelly stock firmware, not mongoose-os. It is being worked on and will be fixed with the upcoming 1.11 release. Thanks for reporting though.

closed 2021-05-19 by kzyapkov

dekatWin, 2023-06-15:
now we have 2023 and i am having the same issu.... so its still not fixed on 1.0 Beta3

```
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
12:54:07
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
12:54:07
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
12:54:07
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
12:54:07
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
12:54:07
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
12:54:07
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
12:54:07
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
12:54:07
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
12:54:07
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
12:54:07
mgos_mqtt_conn.c:555 MQTT0 queue overflow!
```