cesanta/mongoose-os #579: ESP8266 local build - A file in the root directory with the same name as one in the `src` directory is included in the build.

nliviu, 2021-08-30:
How to reproduce:

- git clone https://github.com/mongoose-os-apps/empty
- cd empty
- touch src/foo.c
- touch foo.c
- add `#include "foo.h"` in `foo.c`
- run `mos build --clean --local --platform esp8266`

Result:
```
foo.c:1:10: fatal error: foo.h: No such file or directory
    1 | #include "foo.h"
      |          ^~~~~~~
```

Cloud build is ok, local builds for ubuntu and esp32 platforms are ok.

DrBomb, 2021-08-30:
I don't know why it would work on cloud. But you're not "touching" `foo.h` anywhere on your steps. Perhaps that? The empty app does not include an "includes" entry either. Just some things I noticed by reading the report.

nliviu, 2021-08-30:
I've added `#include "foo.h"` to show that the `foo.c` in the root directory is added to the build process. `foo.c` in the `src` directory is empty.

DrBomb, 2021-08-30:
I see, now it makes sense, my bad

rojer, 2021-08-30:
this has to do with how GNU make searches for prerequisites - first port of call for implicit rules is always the current directory.
i'm not even sure how to fix it... it's been long on my list to move away from the "giant pile of sources" approach, where all resolution is done via a single implicit rule and VPATH that includes all the libs - the more serious consequence of that is that all source files have to have a distinct base name. i'm almost certain this issue will be solved when that one is addressed too, as we'll use a longer stem for our implicit rules that will not match in the current dir.

another option would be to `cd` away from the app dir for compilation, but that's break other uses, i'm sure.

so, i don't see an immediate solution to this, will probably be solved in the future in a bigger makefile refactoring.

closed 2023-01-10 by nliviu