cesanta/mongoose-os #581: Self re-trigger crashes with stack canary or double exception (ESP32)

scaprile, 2021-09-07:
A callback for an event triggering the very same event in order to simulate a main loop causes canary watchpoint resets or core  panics (Double exception); minimum app attached: [crash.zip](https://github.com/cesanta/mongoose-os/files/7122296/crash.zip)

```c
static void rundemo(int ev, void *ev_data, void *userdata) {
  mgos_event_trigger(MY_EVENT, NULL);
}
enum mgos_app_init_result mgos_app_init(void) {
  if(!mgos_event_register_base(MY_EVENT, "Ping-pong demo")) return false;
  if(!mgos_event_add_handler(MY_EVENT, rundemo, NULL))  return false;
  mgos_event_trigger(MY_EVENT, NULL);
  return MGOS_APP_INIT_SUCCESS;
}
```

scaprile, 2021-09-09:
Observing the source code for the mgos_event_trigger() function I see it just calls the handler, no event queuing, so it is clear to me why this is not proper usage of the function and that mOS does not support an event handler re-triggering itself other than for recursive-function-like purposes (and given enough stack space available)

closed 2021-09-09 by scaprile

rojer, 2021-09-09:
indeed, event handlers are executed in-line so you can't re-trigger event from its handler.