cesanta/mongoose-os #593: Adding api_timer.js gave core dump

alnunez, 2022-01-19:
Hello
I'm doing a very simple app in mJS  Ver 2.20.0. with an ESP8266. Just button press detection and prints, nothing more. Just adding at the begining of the file init.js : load('api_timer.js'); gives core dump, something similar to posted here:

https://community.mongoose-os.com/t/any-mjs-code-gives-exception-and-followed-by-core-dump-on-esp8266-release-2-20-0/2358

sctv, 2022-01-23:
Hello,
I tried the standard mJs demo "mos clone https://github.com/mongoose-os-apps/demo-js app4" with a ESP8266ex , and had the same problem of getting a core dump.
 

Here is the output. 
[Jan 23 22:34:27.182] esp_main.c:157          app4 1.0 (20220123-223019)
[Jan 23 22:34:27.182] esp_main.c:159          Mongoose OS 2.20.0 (20220123-223019/2.20.0-g1885c0f) sp 0x3fffff50
[Jan 23 22:34:27.182] esp_main.c:161          CPU: ESP8266EX, 160 MHz, RAM: 52912 total, 48748 free
[Jan 23 22:34:27.182] esp_main.c:165          SDK 3.0.5-dev(81bde69); flash: 4M
[Jan 23 22:34:27.182] esp_exc.c:236           Reset cause: 6 (sys reset)
[Jan 23 22:34:27.182] mongoose.c:15765        Mongoose 6.18, LwIP 1.4.0
[Jan 23 22:34:27.182] mongoose.c:4932         mbed TLS 2.16.11-cesanta1
[Jan 23 22:34:27.182] mgos_vfs_dev.c:72       sfl0: sysflash (), size 4194304
[Jan 23 22:34:27.214] mgos_vfs_dev.c:72       root: part ({"dev": "sfl0", "offset": 32768, "size": 262144}), size 262144
[Jan 23 22:34:27.214] mgos_vfs.c:173          /: SPIFFS @ root, opts {"bs":4096,"ps":256,"es":4096}
[Jan 23 22:34:27.262] mgos_vfs.c:344          /: size 233681, used: 121233, free: 112448
[Jan 23 22:34:27.374] mgos_sys_config.c:470   MAC: 5e:cf:7f:11:67:d9
[Jan 23 22:34:27.374] mgos_sys_config.c:478   WDT: 30 seconds
[Jan 23 22:34:27.374] mgos_i2c_gpio_maste:248 I2C GPIO init ok (SDA: 12, SCL: 14, freq: 100000)
[Jan 23 22:34:27.390] mgos_ota_shadow_lib:116 OTA over shadow initialised
[Jan 23 22:34:27.390] esp_main.c:137          SDK: bcn 0
[Jan 23 22:34:27.390] esp_main.c:137          SDK: del if1
[Jan 23 22:34:27.390] esp_main.c:137          SDK: del if0
[Jan 23 22:34:27.422] esp_main.c:137          SDK: usl
[Jan 23 22:34:27.422] esp_main.c:137          SDK: mode : null
[Jan 23 22:34:27.422] mgos_wifi.c:307         WiFi mode: AP
[Jan 23 22:34:27.422] esp_wifi.c:123          WiFi mode: AP
[Jan 23 22:34:27.422] esp_main.c:137          SDK: mode : softAP(5e:cf:7f:11:67:d9)
[Jan 23 22:34:27.422] esp_main.c:137          SDK: add if1
[Jan 23 22:34:27.422] esp_main.c:137          SDK: dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw
[Jan 23 22:34:27.422] esp_main.c:137          SDK: :192.168.4.1)
[Jan 23 22:34:27.438] esp_main.c:137          SDK: bcn 100
[Jan 23 22:34:27.438] esp_wifi.c:401          WiFi AP: SSID Mongoose_1167D9, channel 6
[Jan 23 22:34:28.333] esp_main.c:137          SDK: bcn 0
[Jan 23 22:34:28.333] esp_main.c:137          SDK: del if1
[Jan 23 22:34:28.333] esp_main.c:137          SDK: usl
[Jan 23 22:34:28.333] esp_main.c:137          SDK: add if1
[Jan 23 22:34:28.333] esp_main.c:137          SDK: dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw
[Jan 23 22:34:28.364] esp_main.c:137          SDK: :192.168.4.1)
[Jan 23 22:34:28.364] esp_main.c:137          SDK: bcn 100
[Jan 23 22:34:28.364] esp_wifi.c:447          WiFi AP IP: 192.168.4.1/255.255.255.0 gw 192.168.4.1, DHCP range 192.168.4.2 - 192.168.4.100
[Jan 23 22:34:28.364] mgos_http_server.c:370  HTTP server started on [80]
[Jan 23 22:34:28.379] mgos_provision_btn.c:87 Factory reset button: pin 0, pull up, hold_ms 0 (hold on boot)
[Jan 23 22:34:28.395] mgos_provision_stat:120 Max state: 0
[Jan 23 22:34:28.395] mgos_rpc_channel_mq:218 0x3fff108c esp8266_1167D9/rpc
[Jan 23 22:34:28.395] mgos_rpc_channel_ua:320 0x3fff15fc UART0
[Jan 23 22:34:28.411] mgos_init.c:33          Init done, RAM: 52912 total, 41944 free, 41952 min free
[Jan 23 22:34:28.442] esp_main.c:210          Stack overflow! Tag 3,1328,0x3fff2de4 ptrs: -76:0x40266e88 -108:0x4024fa51 -124:0x402a70f2 -156:0x4026fda2 -172:0x402a545c -188:0x402a5e85 -252:0x4026fda2 -284:0x402a5eec -316:0x402a54ef -348:0x402a561f -380:0x402a723f -412:0x402a659b
[Jan 23 22:34:28.457] esp_main.c:210          Stack overflow! Tag 3,103,0x3fff1c24 ptrs: -12:0x402a8929 -28:0x40267684 -76:0x40266e8f -108:0x4024fa51 -124:0x402a70f2 -156:0x4026fda2 -188:0x402a5ec2 -252:0x4026fda2 -284:0x402a5eec -316:0x402a54ef -348:0x402a561f -380:0x402a723f -412:0x4
[Jan 23 22:34:28.535] esp_main.c:210          Stack overflow! Tag 3,112,0x3fff0b3c ptrs: 52:0x402a8929 -92:0x402226d2 -124:0x402a90d0 -172:0x4026fda2 -252:0x40106a38 -268:0x4026fda2 -300:0x40106c84 -540:0x402a5f75 -556:0x402a5f75 -588:0x402a61b5 -652:0x402a54ef -668:0x4025fed5 -684:0x4
[Jan 23 22:34:28.535] esp_main.c:210          Stack overflow! Tag 4,112,0x3fff0b3c ptrs: 116:0x4029c9ba 52:0x402683d1 -92:0x402226d2 -124:0x402646cc -140:0x40223954 -188:0x402244ba -252:0x40106a38 -268:0x4026fda2 -300:0x40106c84 -540:0x4021ea43 -572:0x4021dbf0 -588:0x40225d1c -620:0x40
[Jan 23 22:34:28.535] esp_main.c:210          Stack overflow! Tag 3,115,0x3fff0b3c ptrs: 116:0x4029c9ba 100:0x402a663f 68:0x402a66e7 -28:0x4021dbf0 -124:0x402226d2 -140:0x402a8929 -156:0x402a640f -204:0x402a64cf -284:0x40106a38 -332:0x40106c84 -556:0x402a5f75 -572:0x4021dbf0 -604:0x402
[Jan 23 22:34:28.567] esp_main.c:210          Stack overflow! Tag 4,115,0x3fff0b3c ptrs: 116:0x4029c9ba 84:0x4029c9ba 68:0x402a66e7 20:0x402683d1 -124:0x402226d2 -156:0x402646cc -172:0x40223954 -220:0x402244ba -284:0x40106a38 -332:0x40106c84 -572:0x4021ea43 -604:0x4021dbf0 -620:0x40225
[Jan 23 22:34:28.567] esp_main.c:210          Stack overflow! Tag 3,118,0x3fff0b3c ptrs: 4:0x4021ca30 -28:0x40266e88 -60:0x4024fb38 -76:0x402688b7 -124:0x4026fd97 -156:0x402a58c0 -220:0x402244ba -252:0x402a723f -284:0x402a659b -316:0x402a8929 -332:0x402a663f -364:0x402a66e7 -396:0x402a
[... cut ...]
[Jan 23 22:34:31.512] esp_main.c:210          Stack overflow! Tag 2,0,0x3fff374c ptrs: 84:0x4024fb38 52:0x40260808 36:0x402a3a90 20:0x4026fda2 -12:0x402a707e -44:0x402a56fc -92:0x402a5d00 -188:0x402a723f -220:0x402a659b -252:0x402a66e7 -268:0x402a663f -284:0x402a5f75 -300:0x402a66e7
[Jan 23 22:34:31.512]   at init.js:9
[Jan 23 22:34:31.512] MJS error: failed to exec file "api_shadow.js": parse error at line 22: [=== 3 || ]
[Jan 23 22:34:31.526] mgos_mongoose.c:66      New heap free LWM: 28336
[Jan 23 22:34:31.526] mgos_ota_core.c:1589    UID: 0350632c41d101b5, license: none

RobertKoval, 2022-05-22:
I have the same issue with ESP 8266.

tszebeni, 2022-05-31:
Same issue here with ESP8266, not a great experience following the quickstart guide

tonchikk, 2023-02-05:
[core-tsh-esp-js-esp8266-20230205-222034.zip](https://github.com/cesanta/mongoose-os/files/10611942/core-tsh-esp-js-esp8266-20230205-222034.zip)
Here is a coredump with the same:

[Feb  5 22:20:48.951] esp_main.c:157          tsh-esp-js 2023.0.1 (20230205-191210)
[Feb  5 22:20:48.951] esp_main.c:159          Mongoose OS 2.20.0 (20230205-191210/2.20.0-g1885c0f) sp 0x3fffff50
[Feb  5 22:20:48.951] esp_main.c:161          CPU: ESP8266EX, 160 MHz, RAM: 53552 total, 50924 free
[Feb  5 22:20:48.951] esp_main.c:165          SDK 3.0.5-dev(81bde69); flash: 4M
[Feb  5 22:20:48.951] esp_exc.c:236           Reset cause: 4 (soft reset)

[Feb  5 22:20:50.184] mgos_init.c:33          Init done, RAM: 53552 total, 43372 free, 43376 min free
[Feb  5 22:20:50.184] esp_main.c:210          Stack overflow! Tag 3,126,0x3fff168c ptrs: 116:0x40299a9f 84:0x40298dfb 36:0x40298e9f 4:0x40298f47 -28:0x402987d5 -76:0x40298a15 -124:0x40297d53 -156:0x40297e83 -188:0x40297deb -204:0x4029b189 -220:0x40298c6f -268:0x40298d2f -316:0x40299952

tonchikk, 2023-02-05:
On 2.20 it's dumping core, 
If I'll set 2.19.1 version only for mjs - then system is booting but JS stopping working right after first timer's call. Helping to set 

libs_version: 2.19.1 #${mos.version}
modules_version: 2.19.1 #${mos.version}
mongoose_os_version: 2.19.1 #${mos.version}

in main mos.yms (this does not working if I'll add those lines to mos_esp8266.yml )

scaprile, 2023-02-06:
All these posts have two things in common: ESP8266 and "Stack Overflow".
The standard ESP8266 modules do not have enough RAM to run recent mOS versions. Versions 2.19 and below did not trap on  some stack overflows that 2.20 and higher do.

tonchikk, 2023-02-06:
> The standard ESP8266 modules do not have enough RAM to run recent mOS versions.
Sad news I'd say. ESP32 price/gpio_count_per_board value is lower, but I have dozen or two of ESP8266 from 2017 which could be successfully fitted into non-regulated 220v ceiling lights with relays. 
Will stuck on old binaries then and keep updating only JSes...