cesanta/mongoose-os #594: Power Management in ESP32

subhash-coder, 2022-01-24:
Hello, I'm using esp32 with an esp-idf development environment, my project is camera_web_server, where it sends a stream of JPEG images to a browser when I click on the "start stream" button on a web page, so what I need to do is, keep my ESP32 in low power mode until it gets a request from a web browser.

alnunez, 2022-01-25:
Hi. Most of the power consumption of ESP32 came from WiFi. If you are waiting for a web request, you need to keep WiFi turned on all time, so there is no much power you can save at all.

rojer, 2022-01-26:
well, this is not quite true. devices only need to be active occasionally, to maintain association with the AP.
ESP32 has different power-saving modes that make it turn radio off when not needed. bu default it's off (0), but you can set `wifi.sta_ps_mode` to 1 or 2 for different degree of power saving.

subhash-coder, 2022-01-26:
Hello thank you guys for helping, can I use ULP in my project.

scaprile, 2022-01-26:
I don't know if this is the right place (nor if it is my call), if you want to discuss all things your project meet me at [the forum](https://community.mongoose-os.com/). In my case I don't see a server being able to sleep unless the AP has some form of store & forward. The browser's SYN can come at any time and you must be awake to handle it or the IP datagram be queued at the AP until you say hi after the beacon. Unfortunately there's been quite more than a long decade since I was toying with end-devices sleeping on a WiFi network, never with a sleeping server, and that feature might already be built-in on newer APs; IIRC it was not for 802.11b/g and I would risk saying that neither it was for n.

rojer, 2022-01-26:
indeed, deep sleep is unlikely to be possible just because of how disruptive it is, but wifi radio can be put to sleep between beacons, that's what ps_mode does, essentially. there might be slight delay to reacting to connections, yes.

closed 2022-01-26 by rojer