cesanta/mongoose-os #595: ESP32 SNTP sys.tz_spec weird result

alnunez, 2022-01-25:
Hi
I'm working with an ESP32 with mongoose OS V2.19.1.
Added SNTP library and setup and tried tho configure time zone for AMERICA/BOGOTA adding the following line to mos.yml

 - ["sys.tz_spec", "<-05>5"]

Bogota Time is UTC-5, however, after rebooting and SNTP sync the time showed was Greenwich plus 5 hours!!. Just for curiosity, changed the line this time to:

 - ["sys.tz_spec", "<+05>5"]

Rebooted, and SNTP synced. Now the time is correct!. Does somebody experienced the same thing?

scaprile, 2022-01-25:
May I suggest you go to [the forum](https://community.mongoose-os.com/) and search there ? I've answered this at the forum and even have working examples in Spanish

DrBomb, 2022-01-25:
Some questions:

- How did you read the time shown? Because **the SNTP library shows you the unix timestamp on update regardless of your tz_spec value**
- tz_spec only affects a very specific subset of functions. [localtime_r](https://en.cppreference.com/w/c/chrono/localtime) is one of them, while other functions like the widely used `mg_time()` will only return unix time as it should. What are you using?

Time inside C functions, and specially on mongoose can be complicated for begginers. I would suggest, @alnunez, that you make a sample app if you think that indeed, this is a bug on the code.

Good luck!

alnunez, 2022-01-26:
Here is the exact piece of code. A function that builds a message, including datetime.  Timer.fmt() seems affected by tz_spec, because i tried different values, and all of them changed the time zone when printing timestamp.

function buildMsgUl(){

	message_ul.sensor_id=DEVICE_ID;	
	let extTemperatureCstr = roundNdigitsTostr(getTempC(),2);
	message_ul.temperature_ext=extTemperatureCstr;
	let intTemperatureCstr = roundNdigitsTostr((5/9)*(ESP32.temp()-32),2);
	message_ul.temperature_int=intTemperatureCstr;	
	let batVstr = getBatV();
	message_ul.battery=roundNdigitsTostr(batVstr,2);
	let timeunix = Timer.now();
	let timestamp = Timer.fmt('%Y-%m-%d %H:%M:%S',timeunix);	
	message_ul.timestamp = timestamp;
}

DrBomb, 2022-01-26:
mJS is outside my expertise, I don't use it.

That said, here's your reason https://unix.stackexchange.com/questions/104088/why-does-tz-utc-8-produce-dates-that-are-utc8

Basically the POSIX tz formats are 3 letters plus the offset. But the offset has to be on the opposite sign. Hence why `["sys.tz_spec", "<+05>5"]` worked for you. Just use that one.

closed 2022-01-26 by DrBomb

alnunez, 2022-01-26:
Well after some more experimentation i found some discrepancies in tz for America/Bogota as follows:

from: https://github.com/mamuesp/timezones "America/Bogota":**"<-05>5"**
This value doesn't work, gave time UTC+5

from: https://github.com/openwisp/netjsonconfig/blob/master/netjsonconfig/backends/openwrt/timezones.py "America/Bogota", **"COT5"**
This value Works OK as expected (gave time UTC-5 )

alnunez, 2022-01-26:
> mJS is outside my expertise, I don't use it.
>
> That said, here's your reason https://unix.stackexchange.com/questions/104088/why-does-tz-utc-8-produce-dates-that-are-utc8
>
> Basically the POSIX tz [...]

That's the answer i was looking for! thanks again!