cesanta/mongoose-os #605: esp32-c3 build fails with `known CMake command "add_prebuilt_library"`

nliviu, 2023-02-23:
After the latest commit as of 22/02/2023 build for esp32c3 fails
```
CMake Error at /mongoose-os/platforms/esp32xx/esp32xx.cmake:11 (add_prebuilt_library):
  Unknown CMake command "add_prebuilt_library".
```
It looks like `/mongoose-os/platforms/esp32xx/esp32xx.cmake` is missing
```
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
```

rojer, 2023-02-23:
@a2800276 can you check and put up a fix please?

a2800276, 2023-02-23:
Ah yes, didn't copy over to esp32c3, sorry. 
Let me try to set up a mini project to build for both plattforms.

a2800276, 2023-02-23:
Ok, I've copied the CMakeFiles.txt and tested it: https://github.com/cesanta/mongoose-os/pull/606

I'm not sure how you guys typically go about testing the build. I tried building `demojs` app for esp32 and esp32c3 before and after, before esp32 worked and c3 failed, after both build suceeded.

closed 2023-02-23 by rojer

a2800276, 2023-02-23:
No worries, I should have tested the PR before submitting. Sorry about being the elephant in the china shop in the first place :)