cesanta/mongoose-os #611: Mongoose OS end

Younis08, 2023-08-20:
Hi @rojer, more than a year now since we've seen an update to the Mongoose OS!
Has it come to an end? will we see more of it? or that's it?

scaprile, 2023-08-20:
Mongoose OS is one of Cesanta's main products, you might want to contact them directly at cesanta.com
If you have a gitter account:
https://matrix.to/#/!wqkTLOnwOKBomOgPhv:gitter.im/$Jb4PMr9RvHst3nElxA7FCa03PHLVaL7F85-G7clFi8o?via=gitter.im

energiumOn, 2023-08-22:
So, that’s it!!… understood.
It was still my best choice for alternative firmware on Shelly… sad fir that 👍

scaprile, 2024-01-31:
Perhaps you might want to re-read that 
> Mongoose OS is one of Cesanta's main products

and if you have a _specific_ issue, question, or need, 
> you might want to contact them directly at cesanta.com

Otherwise, Mongoose is working perfectly fine as it is.

energiumOn, 2024-01-31:
Yep! That’s the end of it… It is confirmed. Rojer is no longer maintaining the firmware. He is now employed by Shelly. So do not expect any further progress in this project… 👍

scaprile, 2024-01-31:
I don't think @cpq or the rest of Cesanta would agree with that, though.

pimvanpelt, 2024-01-31:
Nor would I as a longstanding member of this community.  I still maintain (and run) Mongoose OS libraries and binaries. 

Your question here is "why is there not an update" -- but have you considered picking up the phone, or sending an e-mail:
Postal: Cesanta, 13 Edward Place, Bloomfield avenue, Dublin 4, Ireland, D04 C2F4
Phone:  +353 1 592 5476
Mail:  [support@cesanta.com](mailto:support@cesanta.com)

closed 2024-01-31 by pimvanpelt

energiumOn, 2024-01-31:
This may not please you, but the main player is gone. Why should we call you to confirm this. It’s obvious and was announced by Rojer himself… RIP 👍