cesanta/mongoose-os #613: mos tool --port option returns "Unknown command"

Ironhydroxide, 2024-06-11:
https://mongoose-os.com/docs/mongoose-os/userguide/mos-tool.md using "--port" flag does not work. 
Returns "Unknown command" when ran. 
Attempted to use port flag to run RPC over MQTT, but always got the Unknown command result. 
Tried simpler useage (local machine) and port option still fails. 
When local port is selected and port option flag is omitted, commands succeed, indicating an issue with the "--port" flag itself. 

Tested with 2 different computers, 2 different OS(windows 10, windows 11), and 2 different Browsers(chrome, Firefox)
all setup according to https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md and cloned the demo-js app. 

> $` mos version
The Mongoose OS command line tool
Version: 2.20.0
Build ID: 20211215-002601/2.20.0-g0278853
Update channel: release
Command completed.
$ mos ports
COM14
COM17
COM18
COM5
Command completed.
$ mos --port COM14 flash
Unknown command
Command completed.
$ mos flash
Port not chosen
Command completed.
$ mos flash
Loaded app1/esp32 version 1.0 (20240611-151204)
...{omitted program loaded logs}
Command completed.
$ mos --port COM14 flash
Unknown command
Command completed.

scaprile, 2024-06-11:
```
$ mos --port /dev/ttyACM0 console
[Jun 11 14:02:12.992] 0  2ongoose.c:53:mg_tcpip_init  MAC not set. Generated random: 0:2d:cf:46:29:04
```
It _does_ work. It may not be working _for you_ under some circumstances.
I suggest you ask for help at the forum https://community.mongoose-os.com/

DrBomb, 2024-06-11:
I've used mongoose for 7 or so years and I've never seen that command response. I usually roll with the `latest` build instead of release. Try it out https://mongoose-os.com/downloads/mos-latest/win/mos.exe

Ironhydroxide, 2024-06-11:
> I've used mongoose for 7 or so years and I've never seen that command response. I usually roll with the `latest` build instead of release. Try it out https://mongoose-os.com/downloads/mos-latest/win/mos.exe

Thanks, tried this... but still get the same result. 

> $ cd C:/mos/problem/app1
Command completed.
$ mos --port COM14 flash
Unknown command
Command completed.
$ mos version
The Mongoose OS command line tool
Version: 202303131403
Build ID: 20230313-145707/2.20.0-20-gb44964e-master
Update channel: latest
Command completed.
$ mos ports
COM14
COM17
COM18
COM5
Command completed.
$ mos --port COM14 flash
Unknown command
Command completed.

Ironhydroxide, 2024-06-11:
> ```
> $ mos --port /dev/ttyACM0 console
> [Jun 11 14:02:12.992] 0  2ongoose.c:53:mg_tcpip_init  MAC not set. Generated random: 0:2d:cf:46:29:04
> ```
> 
> It _does_ work. It may not be working _for you_ under some circumstances. I suggest you ask for help at the forum https://community.mongoose-os.com/

Sadly, the system is not allowing signups.   Clicking the "Create your account" button does nothing. 
I've searched through the community pretty heavily though, to make sure my syntax was correct on the command, and found nobody else who had reported a similar issue.

DrBomb, 2024-06-11:
I've never seen `Unknown command` on mongoose nor on anything trying to use serial ports. Is this not a normal windows instance? Different terminal? Running on a virtual machine? Why does it have a $ for a prompt? And why does it print "Command completed." when the mos binary exits?

Ironhydroxide, 2024-06-11:
> I've never seen `Unknown command` on mongoose nor on anything trying to use serial ports. Is this not a normal windows instance? Different terminal? Running on a virtual machine? Why does it have a $ for a prompt? And why does it print "Command completed." when the mos binary exits?

I have no clue what would be different about it. 
Windows 10 Enterprise, and Windows 11 Pro, Neither running in VM, opened mos by double clicking the mos.exe. 
I am very very new with mongooseos, so could easily be doing something wrong, thus me trying to follow the guides exactly. 

[image]
[image]

DrBomb, 2024-06-11:
OK. That's your problem. Although opening the mos.exe directly opens an interface I do not use it at all. Just run the mos tool via the command prompt. Don't use the GUI that opens.

DrBomb, 2024-06-11:
Also, and sorry for the double comment. The mos tool might linger in the background as I think it doesn't close when you close the window. So make sure to terminate the GUI process if it stays open.

closed 2024-06-11 by Ironhydroxide

DrBomb, 2024-06-11:
> I wonder why the GUI doesn't.... :/

Because it is a relic of their first iterations where they had a code editor and a bunch of functionality that made a lot more sense. Mostly because in that time mOS was banking on mJS a lot. Moving forward mJS was changed to a "prototyping language" and they downgraded the GUI to what I believe is a mostly unusable collection of windows.

Can't blame them though, if anything I'd like the GUI to be completely removed to avoid these issues moving forward, the rest of the mos program is OK.

> I'm getting "not valid JSON String" errors...

JSON strings need to be properly escaped in Windows. String arguments will not accept single quotes `'` as its delimiter so any JSON needs to be enclosed on double quotes `"` and every single internal double quote will need to be escaped with a backslash `\` so your command would be `mos --port mqtt://10.201.102.40:1883/esp32_8ae410 call OTA.Update "{\"url\": \"http://10.12.24.3:3000/firmware/firmware.zip\", \"commit_timeout\": 300}"`

Ironhydroxide, 2024-06-11:
> Because it is a relic of their first iterations where they had a code editor and a bunch of functionality that made a lot more sense. Mostly because in that time mOS was banking on mJS a lot. Moving [...]

Thank you so much for your help!!!