cesanta/mongoose-os #616: Cannot use Mac os to install mongoose OS

itdog1973, 2024-11-14:
Hi, I am trying to install mongoose Os with homebrew on my mac. And I follow the offical install step 

brew tap cesanta/mos
brew install mos
mos

But I got the following app. It is completely different one than I install in my window...

[image]

Could anyone help?

scaprile, 2024-11-14:
Did you try cesanta/mos or however your brew utility needs to be told to get something from a specific repo or tap ?

Perhaps instructions here https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md should be updated for that, as there is some AI app named `mos` now.

itdog1973, 2024-11-14:
Hi scaprile, 

Thanks for helping.

I have tried brew tap cesanta/mos first and I thought "brew tap cesanta/mos" is telling homebrew to add the cesanta/mos repo so that I can then install the cesanta/mos from there ? but I got the wrong mos

And, I also tried "brew install cesanta/mos", but it says " No available formula with the name "mos""

I have tried to register a forum account ,but it seems the forum no longer send the verification email so that I am not able to complete sign up.

May I know which chat you are referring to ?

scaprile, 2024-11-14:
> May I know which chat you are referring to ?

[image]

itdog1973, 2024-11-19:
> > And, I also tried "brew install cesanta/mos", but it says " No available formula with the name "mos""
> 
> I don't use brew, and I don't have a Mac
> 
> I read this:
> 
> > The fully-qualified name of a tapped formula: Sometimes a formula from a tapped repository may conflict with one in homebrew/core. You can still access these formulae by using a special syntax, e.g. homebrew/dupes/vim or homebrew/versions/node4.
> 
> and based on that I would `brew install homebrew/cesanta/mos`

Thanks for your help!

PadLex, 2025-02-16:
What worked for me was `brew install cesanta/mos/mos`

kylefmohr, 2025-09-29:
> What worked for me was `brew install cesanta/mos/mos`

neither of these commands work for me. This one got further:

```
kyle@16:/Users/kyle/mos$ brew install cesanta/mos/mos
==> Fetching downloads for: mos
==> Fetching dependencies for cesanta/mos/mos: boost
==> Downloading https://ghcr.io/v2/homebrew/core/boost/manifests/1.89.0
Already downloaded: /Users/kyle/Library/Caches/Homebrew/downloads/cacc3a33217419abc47ff1b91c4c257843b2e9c6dfd9f3ee769a45fb21eac23c--boost-1.89.0.bottle_manifest.json
==> Fetching boost
==> Downloading https://ghcr.io/v2/homebrew/core/boost/blobs/sha256:eb36169206c88a8e9c921516356b8e48cd2019bae7210511e46f83dc9428760e
Already downloaded: /Users/kyle/Library/Caches/Homebrew/downloads/4044a1a0feaf05237230ba8579fb563cb3e11bf8e75c681ea60074c507de3e34--boost--1.89.0.sonoma.bottle.tar.gz
==> Fetching cesanta/mos/mos
==> Downloading https://mongoose-os.com/downloads/homebrew/bottles-mos/mos-2.20.0.big_sur.bottle.tar.gz
curl: (22) The requested URL returned error: 404

Error: mos: Failed to download resource "mos"
Download failed: https://mongoose-os.com/downloads/homebrew/bottles-mos/mos-2.20.0.big_sur.bottle.tar.gz
```